ControlSystemStudio/phoebus #4: Managing preferences

shroffk, 2017-07-05:
Include the use of the java preferences API

kasemir, 2017-08-23:
Basic API is
```
    // Get preferences for the _package_ of MyClass
    final Preferences prefs = Preferences.userNodeForPackage(MyClass.class);

    // Read, write
    int calls = prefs.getInt("calls", 0);
    prefs.putInt("calls", calls+1);
```

Suggestions:

Each module that needs to read (and in some cases write) preferences uses some class in its base package to access preferences.

The launcher has command line options to export and import preferences, basically calling
```
    Preferences.userRoot().node("phoebus").exportSubtree(new FileOutputStream("export_file.xml"));
```
resp.
```
    Preferences.importPreferences(new FileInputStream("import_file.xml"));
```

This provides the essential functionality.

Downside:
 * The exported/imported settings are in a rather verbose XML format. Not a show stopper, could eventually support export/import in other formats.
 * The default implementation stores the settings in ~/.java or Windows registry, depending on OS. This means there's one set of preferences per user. Each instance of Phoebus for that user will have the same preferences. Same user cannot run instances with different settings. To support this, we would have to implement our own `PreferencesFactory` service.

kasemir, 2017-08-24:
The PV Tree and core/pv library use `Preferences`.
Reading them is easy, and it overall "works", with some concerns:

### Knowing what's available
With RCP, a `settings.ini` in each plugin can list all preference settings, define the defaults, and ideally add some explanatory comments.

The closest equivalent is now for example the `org.phoebus.applications.pvtree.Settings.java` source code.

You can "export" the current settings, but if code only reads and never writes the settings, that export would be empty, not presenting the available preferences and their current (default) values.

I thus coded all access like this:
```
Preferences prefs = Preferences.userNodeForPackage(...);
value = prefs.get(THE_NAME, default_value);
prefs.put(NAME, value);
```

So once the code runs, the preferences are populated, initially with the default values.

When then exporting them to a file, you can at least see the values.
There is no explanation on their meaning in the XML, though.
Still, you can edit the exported file and then import it.

### One set of Preferences per User
Once you run
```
./phoebus.sh -settings settings.xml     
```
this imports the preferences from `settings.xml` for that user.
From then on, you can run without the command line arg.

This can be an advantage, but also a problem if you want to run different instances with different settings, for example different "top displays".
Would have to use different user accounts, or implement our own `PreferencesFactory`.

kasemir, 2017-08-31:
Questions:

- Is there a way to change the default location for Linux and Mac from  ~/.java? That would help on beam lines with shared $HOME.

- Can defaults for a module be provided in property files? This would allow documentation of available settings.

shroffk, 2017-09-11:
Ok there are 2 options

1. create a new preference service which is then used for all phoebus applications, the service wraps around the java preferences but add things like unique nodes for each instances etc...

2. implement our own  AbstractPreferences and Preference Factory
This might be a lot of work but it would mean that all applications can just use the default java preferences api

kasemir, 2017-09-11:
I suggest we certainly stay with the `java.util.prefs.Preferences` API, and maybe eventually create our own `PreferencesFactory`, i.e. option 2.

Looks like for Linux you can already change the 'root' directory via properties like `java.util.prefs.systemRoot` and `java.util.prefs.userRoot`, so it is possible to have more than one set of preferences for a user, which was my main concern.

A lesser concern is documenting the available settings.
If our code only reads preferences, you have no easy way to see what it's trying to read, what the defaults are, and what they mean.
So my code for now also always writes the settings: As soon as you once ran an application, you can 'export' the preferences and thus see their default value. Still missing the explanation.

shroffk, 2017-09-11:
I was looking at the ```FileSystemPreferences.java``` 
I have tried to make the following changes

 1. remove all native calls to unix bindings, I would like it if we have one way of handling these of all systems

2. added a unique instance id, so the each phoebus instance gets its own tree under the user root node.

we cna have a hangout tomorrow to discuss the following
- startup/shutdown hooks and persistence
- application specific preferences

kasemir, 2017-09-11:
Yes, the `FileSystemPreferences` which are the default on Java look like a very good starting point.

I've made progress with handling cmd line args, and optionally maintaining only one instance. Starting phoebus once more from the command line would then open the requested files in the first instance. A hangout tomorrow sounds good. 11:00? 13:00?

kasemir, 2017-09-27:
Comments from discussion that started on #52:

 *  Preferences use a package name, Preferences.userNodeForPackage(SomeTopLevelClassInYouApp.class);

 *  Each app feature has a properties file that lists and explains the default preferences. Maybe call that "preferences.properties" so it's easy to spot? Should include the package name as a comment.

 *  We do NOT write the preferences. So if you export preferences, that's mostly empty.

 * To set preferences, you for now need to locate the "preferences.properties" and construct the XML for import. The Launcher could support a settings import that reads a property file type of format with lines "package/setting=value" because that's MUCH easier to construct than the XML.

kasemir, 2018-06-28:
Preference use is quite established now and documented in http://phoebus-doc.readthedocs.io/en/latest/preferences.html, so I think we can close this issue

closed 2018-06-28 by kasemir