ControlSystemStudio/phoebus #8: How to manage docking

shroffk, 2017-07-17:
@kasemir says"

> 
> Hi:
> 
> Claudio, I think you mentioned in the last hangout that Java 9 would include updated Tab/TabPane/.. support that includes docking.
> Do you have any reference for that?
> The closest I found was a request for docking tabs, ending in "targeted JDK10", https://bugs.openjdk.java.net/browse/JDK-8092098 
> 
> I briefly looked at the following
> 
> http://berry120.blogspot.co.uk/2014/01/draggable-and-detachable-tabs-in-javafx.html
> https://github.com/alexbodogit/AnchorFX
> https://github.com/RobertBColton/DockFX
> 
> The first one has a trivial amount of code. The blog includes a video which just shows a window with tabs, you can drag tabs out into individual windows, then add tabs to them. I think that's perfect, just like a web browser.
> 
> AnchorFX and DockFX seem similar, in fact https://github.com/alexbodogit/AnchorFX/issues/20#issuecomment-263724971 suggests that one copied the other when it should have been a clone.
> They support the idea to 'split' a window area into subpanels, like Eclupse where dropping a tab to the left/right/top/bottom re-sizes the original tab that occupied the complete area. That's the type of operation most CSS users never quite understand and thus hate.
> 
> The draggable-and-detachable-tabs-in-javafx.html demo is nice on Mac OS X, and again I think it does all that we need.
> Trouble is that none of the demos work on Linux. I tried RedHat 6 and 7, with Java 8 and 9.
> Starting to drag any tab results in a tiny new window, but can't 'drop' anywhere.
> For AnchorFX & DockFX there's an addition window with docking icons, I guess to drop  left/right/top/bottom within the existing window, but again I can't 'drop' there.
> 
> -Kay

claudio-rosati, 2017-07-17:
My answer:

> Hello Kay,
>
>your reference is correct. I’ve seen this page probably before it was re-targeted for JDK10. That is very bad.
>
>I like AnchorFX/DockFX, and I prefer them , the problem is that these project seems inactive.
>
>There are also the following:
>
>https://github.com/andy-goryachev/FxDock
>http://www.drombler.org/drombler-fx/0.10/docs/tutorial/docking-framework.html#dockingFramework
>
>Claudio

kasemir, 2017-08-25:
There is basic docking: New Probe, PVTree, ... opens in tab. Tabs can be rearranged, detached.

Main thing missing is a way to get a more 'standalone' view for a Stage that has a single tab, https://github.com/kasemir/org.csstudio.display.builder/issues/271

kasemir, 2017-09-12:
Stages with just a single tab are now 'standalone', hiding that tab.
When another tab is dragged in, the tab headers re-appear.
This does mean that if you have a stage with a single item, you cannot drag that single item out, since it has no visible tab that you can grab and move. Same with web browsers, which thus often have a preference setting to "always show tab".

claudio-rosati, 2017-09-12:
Probably a similar preference should be provided

kasemir, 2017-09-15:
There is now a runtime option to always show the tabs, it's persisted in preferences.

--> All the basic ideas for docking have been implemented.

closed 2017-09-15 by kasemir