ControlSystemStudio/phoebus #12: "Open Files" from menu, command line, file navigator

kasemir, 2017-08-23:
Eventually, we need to support a "File/Open" type of menu, a way to open files from the command line, and a way to open the respective application from the OS-specific file navigator.

This requires a way to associate file names with applications, a way to detect file access from the navigator, and a way to manage phoebus instances.

To associate file types (file extension?) with applications, we could simply extend the `Application` interface. It could for example have a `String[] getHandledExtensions()` method as well as `open(String file_or_url)`.

The file navigator behavior depends on the OS. On Linux, this just means that the command-line launcher needs to accept file names as arguments.

Finally, this requires some way for the launcher to detect an already running instance so that it can forward the request to the running instance. http://aniefer.blogspot.com/2010/05/opening-files-in-eclipse-from-command.html describes how this was done for SWT, which uses different implementations for GTK, Mac and Windows to basically locate the main window by name. At least for our Linux installations this never quite worked once an instance was running for some time. One idea would be a TCP server: Our `PhoebusApplication` starts a TCP server on a fixed port where it accepts `"filenname_or_url [optional args like macros]\n"`. If the port is already in use, we assume that an instance is already running, and send file open requests to the running instance. If the port is not in use, we open the UI and handle received file names. There could be an optional  `-standalone` command line option that skips the TCP server and always opens the UI, in case you do want multiple instances on the same computer.

kasemir, 2017-08-23:
The existing `start()` could turn into `start(String... file_or_url)`, so it's not a new call.

I'm not sure about `URI`, because that is very picky.
Users should be able to call the tool as
```
phoebus "/home/controls/vacuum displays/Main Display.bob"
phoebus "http://some_server/vacuum displays/Main Display.bob"
```

We do not want to require `file://home/controls/vacuum displays/Main%20Display.bob`.

For command line arguments we could pass them through a helper that turns command line args into URI. But them I'm still concerned that URI is somewhat limited to reading. You can get an input stream from a URI. But one application would be the display builder editor, which wants to write. So it would have to check if the URI is a file, then get the file for writing, .. ==> Might as well just pass a string and the application can then decide to use that as a File, URI, ..?

shroffk, 2017-08-24:
I feel we can keep the start as it is and have a separate open method. The start can do resource allocation, register services, add listeners, etc... it will would be associated with resources shared by multiple instances of the application (like the connection of various logbook views, or common resources for many bob screens)
The resources of the start would be cleaned up by the stop().

The open can be used by _action_ to launch some ui or execute some action in headless applications.

The different cases of application launching that I can think of
```
open()
```
Default: just open the application views
```
open(String... filePath_URL)
```
Open application with some input file, like opi editor or runtime with a .bob file, open databrowser with a .plt file
```
open(Selection selection)
```
Open an application with some input data, open probe or pvtable with a list of all the selected pv's

kasemir, 2017-10-02:
Opening files from the OS's file browser works, at least for RedHat Linux / Gnome / Nautilus.

The registration of phoebus as an application and associating that with certain files is a bit involved, but possible as described on http://phoebus-doc.readthedocs.io/en/latest/running.html .

Once the installation is done, clicking on a *.pvs file opens the initial Phoebus instance and then the PV Table in ~3 seconds.
If phoebus was already running, clicking on a *.pvs file in the file navigator opens the file in Phoebus within a second.

kasemir, 2017-10-02:
For Mac OS X, tried to use a 'phoebus' shell script to start the product similar to linux, but wrapped as a Mac app.

Idea 1: Create Phoebus.app/Contents/MacOS/Phoebus as per https://gist.github.com/oubiwann/453744744da1141ccc542ff75b47e0cf#file-appify-sh
Can also add css.icns.

That "works" for starting Phoebus from Finder, Dock, ..
But no way to pass arguments. Dropping an example.pvs onto the App icon will just start Phoebus, it doesn't receive the path to the dropped resource.

Idea 2: Use Automator
Start Applications/Automator.
Add "Run Shell Script", select "Pass input: as arguments", paste the content of the phoebus start script.
Save as "Phoebus.app".

That "works" for starting Phoebus from Finder, Dock, **and** can drop example.pvs onto the app icon to start phoebus and then open a PV table for the resource.
**BUT**: Automator-generated app only allows one instance. Drop file three times onto icon, and you don't get phoebus with the 3 files opened. Instead, phoebus is opened and receives the first file. Close phoebus, and it's opened again with the second file. Close once more, and it's opened with the 3rd file.

kasemir, 2017-10-03:
The 'correct' way to handle received file resources on Mac OS X would be by implementing `handleOpenFilesAction` as shown on https://stackoverflow.com/questions/29101472/pass-parameters-to-javafx-application-by-double-click-on-file.
This, however, requires access to `com.sun.glass.ui.Application`, which the Java 9 module exports to some other internal modules, resulting in compilation error
```
[javac]   (package com.sun.glass.ui is declared in
 module javafx.graphics, which does not export
 it to the unnamed module)
```
unless we add `-XaddExports:javafx.graphics/com.sun.glass.ui=ALL-UNNAMED`.

==> Don't support that for now. Mac OS X users will need to invoke File/Open to open additional files, or invoke phoebus on the command line to send additional `-resource` requests to a 'server' instance.

kasemir, 2017-12-05:
Status:

Opening files from the command line as `-resource ...` works well for local as well as remote files.
Can select specific application (display _editor_ vs. _runtime_).
Command line `-server` option allows opening only one instance of the product, which then receives the `-resource` requests.

Opening resources from File/Open as well as a top resources toolbar item is done.

Missing is the integration with OS-specific file browser on Windows, Mac, Linux.
For now we give up on that, because variants of Linux are too different, Mac OS requires internal API, overall seems too much effort.
Might instead consider more support for dropping files, because drag/drop of file names is universal.
Display Editor already accepts dropping image files (will create Picture widget) or display files (will create Embedde Display widget). A "Drop file here to open" type of drop zone would be easy to implement and function universally on all OS.

closed 2018-06-12 by kasemir