ControlSystemStudio/phoebus #22: Managing the migration from cs-studio to phoebus

shroffk, 2017-09-06:
Technical Issues

User Experience/Branding issues

@willrogers @kasemir @berryman4

willrogers, 2017-09-07:
I was previously considering a partial transiton from BOY to Display Builder within CS-Studio.

If Display Builder is not going to be supported in CS-Studio after a period, this is a risky approach.  BOY won't be supported in Phoebus.  So moving from CS-Studio+BOY to Phoebus+Display Builder would have to be done in one big transition.

willrogers, 2017-09-07:
Despite all of Eclipse's limitations, we have now got a nice way of saving and sharing layouts using perspectives at DLS.  I'd like to be able to do a similar thing in Phoebus.

kasemir, 2017-09-07:
I believe you that an EDM to XXX transition is no fun. Are you now 100% on BOY displays in the DLS control room?

Of the 20+ SNS beam lines, 13 have been converted to EPICS. Of the 13, 4 use the display builder. All can be viewed with the display builder, but some details of local operation tied to scripts still require BOY. The remaining beam lines to be converted to EPICS will use the display builder, and those that are still on BOY will also have their scripts transitioned. The BOY -> Display Builder conversion is much simpler than going EDM -> XXX.

For beamlines, I need display builder, data browser, PV Table, scan client UI, alarm UI, logbook & email submission as essential applications. Apart from scan and alarm UI, these are all generic applications that most everybody would use. Probe, PV Tree, job progress UI, Web Links, logbook viewer, launcher for external application would be nice to have.

If Kunal and I continue at the current pace, I think we can have the essential framework (docking panels w/ and w/o 'input', jobs, preferences, persistence) in place by end of this year, and about one year from now the essential applications for SNS beam lines can be ported, where I will likely concentrate on those essential for SNS while Kunal would concentrate on what's essential to BNL and so on. Once phoebus is sufficient for SNS beam lines, I would naturally devote less time to maintaining the non-phoebus code. DLS and others might prefer to stay with the existing CS-Studio source code for longer, but I honestly don't see how you'll be able to do much beyond bugfixes. For BOY, the core tool, the way it's tied to GEF and thus SWT and the UI thread is a dead end.

kasemir, 2017-09-07:
Regarding configuration files:

Application files for Display Builder, PV Table, ... are exactly the same.

Preferences are different: plugin_customization.ini and external, maybe auto-created diirt config files are replaced by, for now, the XML format used by java.util.prefs.Preferences's `exportSubtree`. We might eventually use our own PreferencesFactory, see #4.

Layouts: We thought about a 'memento' type mechanism to persist and restore state. The framework would handle persisting and restoring the windows, tabs, inputs-to-tabs, and each application can then store additional information (like PV Tree could store the PV). This would allow the original Eclipse restart-as-you-left-it behavior as well as start-in-a-known-layout behavior (by basically making the desired state read-only, so it's not overwritten on shutdown).
There have been no discussions about switching layouts at runtime similar to perspectives.
That would be a new issue for discussion: Does anybody need to switch between layouts? Do the hidden panels remain in memory, as in Eclipse, or should they close down and then restart when re-opened?

willrogers, 2017-09-07:
> Are you now 100% on BOY displays in the DLS control room?

We will be by the end of the year.

> For BOY, the core tool, the way it's tied to GEF and thus SWT and the UI thread is a dead end.

No doubt, but if we can't make the transition then our options are two dead ends.

claudio-rosati, 2017-09-07:
I like the idea of a perspective-like solution

kasemir, 2017-09-07:
I can see a use for switching between a few saved layouts at runtime.

What I don't like about the full Eclipse perspective mechanism:
 - Special handling of central 'editor' pane
 - Part stacks that are initially empty, then grow at the expense of other part stacks when a part is added
 - little control over the part stack used for a new pane
 - On restart, tabs that aren't actually active simply show the label, no code runs. We have problem with the alarm "annunciator" which may show up as a tab, but since it's not the active tab, there is no alarm annunciation.
 - When closing a tab, you don't quite know if the code behind it actually stops. It only stops when the part with that ID is actually closed in all perspectives, and even then it may linger around for some time just in case you might open it once more. Until we figured that out, closing the alarm annunciator meant you might still get annunciations for some time

So if we do layout switches, I'd implement that as a complete close-everything, then open the new layout. Of course in the same VM, so it's much faster than opening a saved layout from scratch, but still a complete close-and-open, no gimmicks about keeping hidden things running.

shroffk, 2020-06-30:
I think this Issue tile is much broader than initially listed.

This issue can be said to be about layout... we have a current mechanism in place and I feel this issue can be closed. Future features and bugs can be addressed in new issue tickets.

closed 2020-06-30 by shroffk