ControlSystemStudio/phoebus #32: ResourceHandler service

shroffk, 2017-09-12:


shroffk, 2017-09-12:
TODO:
Update the Applications interface
 - rename Application to AppDescriptor
 - remove the openWithSelection method (this should be handled but the contextMenu contribution)
 - move the open(resource... resource) and all other resource related methods to the resource handler
 - the only methods in the AppDescriptor would be open() and a default open(String... args)
 - have a methods to get display name and app name

shroffk, 2017-09-14:
@kasemir 
I like the direction of the last commit, it mainly used core java libraries
java.net, java.nio, etc...

I am still ironing out the details but I feel that this can help us support all the above operations in a nice standard way.

shroffk, 2017-09-20:
Can you think of more test cases for resources location
relative paths, absolute paths, ... I think we should add more unit test for the utility class

kasemir, 2017-09-20:
It would be good to have tests for
```
parseQueryArgs(createAppURI("some_app"));
parseQueryArgs(createAppURI("some_app?file=x"));
parseQueryArgs(createAppURI("some_app?pv=x&pv=y"));
```

Related: I'm not sure we're "done" with this API.

Right now `AppResourceDescriptor.create(String resource)` receives the complete "some_app?file=x".
This means that `PhoebusApplication.openResource(path_to_resource)` which used to just call `app.create(path_to_resource)` now needs to call that as
`app.create(app.getName() + "?file=" + path_to_resource)` .

Otherwise `AppResourceDescriptor.create(String resource)` needs to have more code to decide: Was I passed a URL where I have to parse the query, or is it just a plain path?

Maybe it's fine as is, `AppResourceDescriptor.create(String resource)` will always get the URL, so then we should change it to `AppResourceDescriptor.create(URL app_resource)`.

In any case, I need to update the way arguments are passed to an existing phoebus instance. That used to only send resource names. Maybe it should send the complete "-app ..." or "-resource ..." command line text, so when the 'server' instance receives additional requests from another command line they're parsed the same way as actual command line arguments.

kasemir, 2017-12-05:
This is done:
```
./phoebus.sh -help
...
Command-line arguments:

-app probe                              -  Launch an application with input arguments
-resource  /tmp/example.plt             -  Open an application configuration file with the default application
..

The '-resource' parameter can be a URI for a file or web link.
The schema 'pv://?PV1&PV2&PV3' is used to pass PV names,
and the 'app=..' query parameter picks a specific app for opening the resource.

Examples:
-resource /path/to/file                                                    - Opens that file with the default application.
-resource file:/path/to/file                                               - Same, but makes the 'file' schema specific.
-resource http://my.site/path/to/file                                      - Reads web link, opens with default application.
-resource file:/path/to/file?app=display_runtime&MACRO1=value+1&MACRO2=abc - Opens file with 'display_runtime' app, passing macros.
-resource pv://?sim://sine&app=probe                                       - Opens the 'sim://sine' PV with 'probe'.
-resource pv://?Fred&sim://sine&app=pv_table                               - Opens two PVs PV with 'pv_table'.
```

Suggest to close the issue.

closed 2017-12-05 by shroffk