ControlSystemStudio/phoebus #48: Access service

claudio-rosati, 2017-09-21:
One of the strong points the few CS-Studio detractors have is that it is not possible to keep the application open and having users able to login/logout inside CS-Studio. 

I think that an access service could very useful in the core Phoebus connected with the preferences (and window/tab manager). 

Including it from the beginning (with a no-op implementation) would save a lot of pain and work in the future.

kasemir, 2017-09-21:
CS-Studio allows login/logout just fine, using LDAP, Kerberos, anything supported by JNDI.

The alarm UI uses that to check if you may change the alarm configuration.

The problem is that Channel Access doesn't allow changing the 'user' of a channel, so even if CSS knows who you are, CA doesn't care.

kasemir, 2017-09-21:
So going with the general idea to prefer core Java functionality over external libraries whenever possible, we should by all means use JAAS as before to authenticate a user.

Feel free to implement what we already have in CSS:
We start with the current user as reported by the OS, but there's a login dialog that uses JAAS so you can authenticate as a different user etc.

The bigger problem will still be: How to use that for what people really want, i.e. control Channel Access or pvAccess write access? In addition, consider the gateway. You're some user in the office, and your tool talks to a gateway. That gateway talks to the IOC. You don't have write access. Now you authenticate as a 'root' user with write access -> How does that information go through to the IOC?

shroffk, 2017-09-21:
> So going with the general idea to prefer core Java functionality over external libraries whenever possible, we should by all means use JAAS as before to authenticate a user.

+1 JAAS would be the best place to start

claudio-rosati, 2017-09-22:
From my understanding, the idea is to have a set of "similar" users (in term of access right) using the same machine in shifts. So for the CA the user is the common login ar the Client computer. The challenge here is to have the "workspace" (let me use this term for now) updated when an user log-in CS-Studio, with his own preferences (color & font files, position of windows and tabs, etc).

About the technology, JAAS is fine for me too.

pavel-ch, 2020-02-12:
Seems that this topic is silent for a while...
Unfortunately I do need an own Authorization implementation. To enable site specific implementation without too much hassle, I think it is enough to simply add the following method to the AuthorizationService class:
```
    public static void init(Authorization auth)
    {
        instance.set(auth);
    }
```
enabling site specific code to supply own Authorization interface implementation at a suitable point.
Is that viable? Or does this break any intentions or security rules?

kasemir, 2020-02-12:
Right now the `Launcher` creates a `PhoebusApplication`, which calls `AuthorizationService.init()`, and that in turn sets the instance to a `FileBasedAuthorization`.

You'd need some way to disable that, because otherwise, based on timing, you won't know which implementation is active in the end.

pavel-ch, 2020-02-12:
Right. Not an issue for me because of using a custom `main` and `PhoebusApplication` equivalent. But in general this is can be a risk of race condition. Maybe instead of disabling `FileBasedAuthorization`, add something like `boolean isInit() { return instance.get() != null; }` method to see if the `instance` is already set, to enable the site specific code to schedule its init to later (probably via slow polling from a threadpool thread)? Advantage: minimal changes in core, most logic in site specific code, no risk for `FileBasedAuthorization` functionality.

kasemir, 2020-02-12:
If you're using your own startup code, then I think your original suggestion is fine for now, i.e. add a new `init(Authorization)` that you can call. The general `PhoebusApplication` keeps calling the existing `init()`.

To somewhat prepare for later extensions, please use something like this in either one instead of a plain `instance.set()`:

```
  if (instance.getAndSet(auth) != null)
     logger.log(Level.SEVERE, "Authorization is initialized more than once");
```

geniejhang, 2020-12-04:
This is wrong request due to my wrong commits. Please ignore it.

closed 2022-03-01 by shroffk