ControlSystemStudio/phoebus #51: 'Help' menu

kasemir, 2017-09-23:
If phoebus-doc is checked out in parallel to phoebus source code, ``Product`` build should include the doc in html form as well as the javadoc of the source code.

From 'help' menu, these are then accessible via a web browser.

Could either display the content in a Stage & WebView, i.e. inside phoebus, or use `javafx.application.Application.getHostServices().showDocument("path/to/phoebus/help/index.html")` to display in the platform web browser (Safari, Firefox, ...).

External web browser has the advantage that it's a full web browser. Users can access any other web pages, download files, play video, .. without using JVM CPU & memory. But especially on windows we could just get a "cannot open web browser" error message.

Embedded web browser will always open up, but the web browser will use JVM CPU & memory, and its features a a little limited.

berryma4, 2017-09-23:
Although help pages are usually simple, it's worth noting that the Java webkit hasn't been kept up to date in Oracle.  I'm not sure if this will improve in openfx.  It's worth keeping an I on.

kasemir, 2017-09-26:
Plan:

 - [ ] Place help for applications into the source code for the application, like phoebus/applications/the_app/doc, with a top-level index.rst. So the material for the manual section of a specific application feature is kept with the source code for that application. 
 - [ ] Have phoebus-doc fetch all phoebus/applications/**/doc/index.rst files, auto-creating a top-level applications page that links to all of them. This gives us an overall 'manual' that includes sections for each application feature. 
 - [ ] When building phoebus product, if the phoebus-doc/build/html is available, copy that into product/doc. 'Help' menu opens web browser (likely the platform web browser) with product/doc/html. This way, the manual is also available inside the product, no network access required.

kasemir, 2017-09-26:
The phoebus 'docs' branch has doc/index.rst files for PV Table, Tree, PV List, Log Config, ..

The phoebus-doc/source/conf.py checks for a ../../phoebus folder with sources. If not found, it calls git to clone the phoebus sources.
Thus having the sources one way or the other, it then includes the application docs in the manual.

That "works" even with readthedocs, except readthedocs would sometimes fail in its initial `pip install` phase (http://readthedocs.org/projects/phoebus-doc/builds/6029866), or at other times spend 15+ minutes in the initial `cloning` phase (http://readthedocs.org/projects/phoebus-doc/builds/6029879/), which look like general network problems unrelated to this.

kasemir, 2017-10-19:
There is a 'help' menu entry which shows either the online help from http://phoebus-doc.readthedocs.io or the bundled help files when the product is assembled via `ant dist`.

Information for contributing to the help content is on http://phoebus-doc.readthedocs.io/en/latest/help_system.html

closed 2017-10-19 by kasemir