ControlSystemStudio/phoebus #55: File vs. URL problem on Windows

kasemir, 2017-09-27:
Trying to write a PV table fails on Windows because C:\some\path doesn't convert into a URL:

[image]

shroffk, 2017-09-27:
hmm...is it possibel to get the stack trace with source line numbers

kasemir, 2017-09-27:
Well, I have "C:\Users\.. " as a file name and call createResourceURL(String).
In line 42 it calls URI.create(), which fails because "C:\Users\.." is not a valid URL.

What follows in createResourceURL() is the right idea: If it's no URL, try Paths.get("C:\..") which would work.
But we never get there because we fail in line 42.

I tried to setup development on a Windows computer, but ran into problems installing the Java 9 support,  #57. So I'm limited to "ant dist" on Linux, then copy the ZIP to Windows, unzip, run.
.. which is still 50 minutes faster than building CSS with maven ;-)

kasemir, 2017-09-27:
I get the file name from prompting the user via org.phoebus.ui.dialog.SaveAsDialog.promptForFile.
So I know it's a `File`, and I call this in the ResourceParser:

```
  public static URL getURL(final File file) {
        return createResourceURL(file.getAbsolutePath());
    }
```

Since it knows it's a file, maybe it shouldn't convert to String only to then fail parsing that String as a URL.
Go directly from File to URL:

```
  public static URL getURL(final File file) {
        return file.toURL();
    }
```

kasemir, 2017-11-20:
Fixed with URL/URI update

closed 2017-11-20 by kasemir