ControlSystemStudio/phoebus #57: Java 9, Eclipse Oxygen

kasemir, 2017-09-27:
Oracle JDK 9 was just released, and Phoebus works fine with that.

Eclipse Oxygen 4.7.1a is to be released on October 11th, 2017.

Until then, need to install the "Java 9 Support for Oxygen" (no longer "BETA").
I've had several issues with the J9 support.
Right now it fails to install in a new 4.7.1 setup on Windows because it cannot download some dependency.

The older version 1.1.1.v20170627-0530_BETA_JAVA9 works fine, but when using 	Help, Check for Updates to update it, I get strange compiler errors where for example
```
String[] strings = ...
for (String element : strings)
```
would complain that 'strings' cannot be iterated. A Project, Clean resulted in the same error.
But just adding a space, deleting the space and saving the file to trigger compilation of just that file fixed the compiler error?!

==> Check if Oxygen 4.7.1a, once released, works out of the box.

kasemir, 2017-10-09:
Tried Oxygen 4.7.1a RC 1.
Added JDK 9+181 to the "Installed JREs", set Eclipse Java, Compiler, Compiler compliance level: 9 for the workspace.

Get
```
Internal compiler error:
java.lang.NullPointerException at org.eclipse.jdt.internal.compiler.lookup.BinaryModuleBinding.create(BinaryModuleBinding.java:64)
```
for `org.phoebus.applications.email.ui.SimpleCreateController.java`.
Looks like Eclipse bug https://bugs.eclipse.org/bugs/show_bug.cgi?id=525522 .

Removed email project from workspace (also removed it from the 'product').
No compiler errors, but whatever JUnit test or 'main' I try to run ends in 
```Class not found org.phoebus...{ThatMainClass}```

==> Eclipse Java 9 support can only get better.

kasemir, 2017-10-11:
Tried Oxygen 4.7.1a , and it doesn't "just work".

The email compiler error may be related to dependencies/.classpath containing an absolute path to like this:
```
<classpathentry kind="lib" path="target/lib/javax.mail-1.6.0.jar"
  sourcepath="C:/Users/Kunal Shroff/.m2/repository/com/sun/mail/javax.mail/1.6.0/javax.mail-1.6.0.jar"/>
```

Even though the workspace source code level was set to "9", many sources showed compiler errors until I set "Java Compiler: Enable project specific settings" and selected "9" there as well.

Finally I removed all the compiler errors, but at runtime, i.e. launched from within Eclipse, anything that needs dependencies cannot find them.

Do all dependencies need to be wrapped as Java 9 modules? Can't use automatic/unnamed modules any more?

kasemir, 2017-10-11:
Looks like Eclipse Java 9 support is plain broken; really disappointing.

The original approach where the `dependencies` project exported all the jar files, and other projects then depended on the `dependencies` project, no longer works at runtime. At compile time, things look good, but at runtime, the code in the dependencies is not found.

Created a J9 Demo project where I try to use the VTypes by importing them as a module:

 1) In the `.classpath`, add `/dependencies/target/lib/vtype-3.0.0.jar` to the module path. Since the vtype.jar isn't really a module (no module-info.java), it's used as an automatic module.
2) In my project, I can now create a module-info.java which `requires vtype`.
3) The code actually runs, as evidenced in the console

--> Great, we can import all the dependencies as modules, even if they don't include module-info files, yet, and the code runs.

*But* Eclipse reports compiler errors, claiming it cannot find the packages from the module.

[image]

kasemir, 2017-10-13:
The "sourcepath="C:/Users/Kunal .." for the java.mail dependency is _not_ the reason for the internal compiler error in org.phoebus.applications.email.ui.SimpleCreateController.java. Still don't know what to do about that, for now excluding email.

For the 'product' to find the jar files that are supposedly exported by the 'dependencies' project, I had to add all the jar files directly as dependencies to the 'product' project, see 9dbfacb .

shroffk, 2017-10-16:
oops...my bad about the hard coded stuff.

Can we have a hangout about this...I am sorry but I am bit confused about this.

kasemir, 2017-10-19:
@shroffk: Tried the latest Java 9.0.1+11 and Eclipse 4.7.1a and found a little more detail on the 
```
Internal compiler error:
java.lang.NullPointerException at org.eclipse.jdt.internal.compiler.lookup.BinaryModuleBinding.create(BinaryModuleBinding.java:64)
for org.phoebus.applications.email.ui.SimpleCreateController.java.
```

This is related to `javax.activation`.
When commenting the code section that adds attachments to the email via activation.FileDataSource, there are no internal JDT errors.

It indeed looks like Eclipse bug https://bugs.eclipse.org/bugs/show_bug.cgi?id=525522 :
"... replacing a JRE module with a non-modular variant of the same classes"

Java 9 includes `javax.activation.DataSource` in `jdk-9/jmods/java.activation.jmod`, and we add another copy in  `dependencies/target/lib/activation-1.1.jar`

shroffk, 2017-10-20:
> Java 9 includes javax.activation.DataSource in jdk-9/jmods/java.activation.jmod, and we add another copy in dependencies/target/lib/activation-1.1.jar

I agree, we should try to use what is part of the jdk first.

shroffk, 2017-10-20:
Also...
I need to update my development environment. I am still using the pre-release one...
I think I can try to update the readme too...is there anything exotic I should be aware of?

kasemir, 2017-10-20:
I've updated the Readme, see https://github.com/shroffk/phoebus/pull/70

kasemir, 2017-10-20:
The remaining unresolved problem with Eclipse 4.7.1a is the email 'attachment' code, where the javax.annotation code from the jar used by the javax.mail.jar conflicts with the javax.annotation code from the JDK9 module. For now I've commented the attachment code.

kasemir, 2018-06-19:
Java 9/10 and Eclipse 4.7.3a work fine.
The email issue is resolved with #191

closed 2018-06-19 by kasemir