ControlSystemStudio/phoebus #64: In memory logbook with markup support

shroffk, 2017-10-04:


shroffk, 2017-10-04:
Markdown Writer FX (https://github.com/JFormDesigner/markdown-writer-fx)

mvitorovic, 2017-10-04:
Java generic markup parsing library 
https://github.com/atlassian/commonmark-java

mvitorovic, 2017-11-14:
We need to make some progress on this. IIRC Caludio was said he will do the initial work?

claudio-rosati, 2017-11-28:
Despite my intentions, phoebus is now a very low priority task for me, being we are preparing for commissioning having a better Display BUilder (meaning also courses etc.) requires me on it almost at 100%

shroffk, 2018-01-29:
The markdown-writer-fx is not supported with java9.

shroffk, 2018-06-29:
Creating some prototype controls for displaying logentries

Single log entry:
[image]

Log Entry Table:

[image]

shroffk, 2018-07-02:
Ok
The time and Title can be in the same line, the body remains as is.
We can also set the Title to bold.

shroffk, 2021-03-09:
commonmark support has been added to the olog ui clients.
currently I don't see a way to add it to the common ui without effecting other users - elog/sns logbook
closing issue

closed 2021-03-09 by shroffk