ControlSystemStudio/phoebus #67: Launching application

claudio-rosati, 2017-10-07:
Evaluate the possibility of use [FXLauncher](https://github.com/edvin/fxlauncher), an "auto updating launcher for JavaFX Applications".

kasemir, 2017-10-07:
Related to #12, opening files from the command line.

At this time, integration with the Linux file browser "works", end users can double-click on a *.pvs file to open the PV Table, re-using an already running instance if available. The end user doesn't need to see the laborious installation steps, but they are still, well, laborious. And the corresponding handling of open-file-events for Mac OS X is not done. Finally, haven't looked at Windows.

Maybe FXLauncher includes handling of arguments?

claudio-rosati, 2017-10-07:
We have to check it. I’ll try to use it in the next days and see what it offers

kasemir, 2020-03-21:
https://openjdk.java.net/jeps/343 : Java 14 includes `jpackage` which generates 'native' formats like msi and exe on Windows, pkg and dmg on macOS, and deb and rpm on Linux.
Evaluating and using something included in the JDK would be preferred over external tools.

georgweiss, 2020-03-21:
That's good news. I will explore it, hopefully very soon.