ControlSystemStudio/phoebus #68: updated vtypes

shroffk, 2017-10-09:
Starting from the org.diirt.vtypes we create the org.phoebus.vtypes (temporarily)
Goals:
use these types in the new connection layer
use these types to be the default mechanism to talk between applications and cs-stuido
use java8 features
incorporate the changes recommended by gabriele in the org.diirt.vtypes.next

shroffk, 2017-10-09:
@kasemir @berryma4 @willrogers 

0a69621
I think this might be nice...we can include the factory methods in the interface itself.

If this is fine, I can update the other classes..

kasemir, 2017-10-09:
I agree. Just like the Java 9 `List` now contains `List.of(..)` to create an instance from the interface.

shroffk, 2017-10-11:
@kasemir @berryma4 @willrogers 
Can we merge this?
I think it is a pretty nice starting point - it has everything the diirt types do and some improvements

kasemir, 2017-10-11:
Yes, merge. It'll only be next week that I can start to use it, though.

kasemir, 2017-10-19:
My next steps:

*  [ ] Adjust build.xml to compile phoebus.vtype
*  [ ] Update phoebus.vtype.pv to use it (instead of external diirt.vtype)
*  [ ] Check phoebus.vtype javadoc for remaining references to diirt

kasemir, 2017-10-19:
Got this warning:
```Note: /home/ky9/git/phoebus/core/util/src/main/java/org/phoebus/util/config/ServiceLoaderOSGiWrapper.java uses or overrides a deprecated API.
```

-> I assume we can simply remove the ServiceLoaderOSGiWrapper

shroffk, 2017-10-20:
yes...yes...
I have never been so happy when removing something

kasemir, 2017-10-20:
I've removed it, which also triggered another travis build of #70, but it still fails with  "ERROR 404: Not Found. download failed. Oracle JDK 9 is NOT installed."

shroffk, 2017-10-20:
With the jdk 9 release we no longer need the post installation step...I updated the travis.yml file accordingly and it seems to work now.
https://github.com/shroffk/phoebus/pull/71

kasemir, 2018-12-17:
We're using the EPICS 7.0.2 vtype and will continue to update to the latest one.

closed 2018-12-17 by kasemir