ControlSystemStudio/phoebus #76: Start/End Date/Time UI

kasemir, 2017-10-30:
We need a Start/End Date/Time UI, i.e. a way for users to enter start and end times for log book retrieval, time-based plots, archive data retrieval.

Conceptually, it needs to be like the org.csstudio.apputil.ui.time package:

 * Some way to hold absolute as well as relative start and end times, i.e. both "from 2017/01/17 13:45 to 2017/02/26 17:27" and "from 3 weeks ago until now" or "from 5 days ago until 2 days ago".
 * Ideally, there's even a way to express "last shift", which would be something like "from 18:00 yesterday to 06:00 this morning";
 * A way to format as text and parse from text, both absolute ("2017/01/17 13:45") and relative ("now", "-4 weeks 2-hours") and that mixed scenarios for "last shift" ("-1 day, 18:00"?). The text representation is used both for serialization and for direct user input.
 * With UI for people who don't want to learn the text-based syntax

In contrast to org.csstudio.apputil.ui.time, it should

 * Use java.time.Instant, Duration etc. as much as possible (unclear if that's possible for relative time)
 * Use Java FX for UI

shroffk, 2017-11-14:
I guess the this would be in the core-ui module

Also, would it be ok to include additional dependencies? like jfxtra

kasemir, 2017-11-14:
Does that come in a version ready for J9?

shroffk, 2017-11-15:
I will have to test it...but first I wanted to figure out where best to put this.

kasemir, 2017-11-15:
core-ui is certainly a good place.
I worry about the long-term liability for added dependencies beyond those that we absolutely need.
Control system communication like CA, pvAccess, mqtt, .. can't be avoided.
We also depend on jython to support scripts, they're critical for our operation.
Mail support is also needed.
Widgets tend to follow fashion trends, are likely to be abandoned.

shroffk, 2017-11-15:
Actually I tested the jfxtra date time picker...and have decide not to use it.
I will try to work with the javafx date picker and see how I can enhance it...and add the necessary time selection options.

shroffk, 2017-11-15:
So after reviewing what we already have  I notice that we have point 1 and point 3 already implemented in different classes.
o.p.util.time.TimeParser (which has support for the "last" and "ago" format)
o.c.apputil.RelativeTimeParser (which supports - n years, etc...)

1. Do we really need both? it just means more formats to handle and more syntax's to be memorized by users.

2. should we support months in relative time definitions, since there are months with 29,30,31 days in the TimeParser I choose to support weeks instead of months since they are always fixed.

kasemir, 2017-11-16:
No, we don't need multiple APIs.

Note, though, that the relative time is based on calendar day, month, year.
"-1 month" doesn't mean "-30.5 days". Instead, it subtracts 1 from the current month, so it's the same day-of-the-month and wall clock time as "now", one month ago, no matter if that month has 28, 29, 30 or 31 days.
Likewise, "-1 day" is not "-24 hours" but same wall clock time, a day ago. That's 24, 23, or 25 hours for normal days vs. DST transitions. Adding support for "-x weeks" is fine, which should then be "-7 days" and again use "day" in the described sense, not just "-7*24 hours" because some days have 23 or 25 hours.

As for the exact syntax, we do need to parse existing Data Browser config files.
If we use a totally different format, we'd need a converter from the old format, so in the end we'd still implement two formats, the "old" and the "new".
I'd prefer having just one format.
Doesn't have to be 100% the old Data Browser format, but it should handle the most common case of "-X hours -Y minutes -Z days", so that plots that used to start with a width of a few minutes, hours, days show up mostly the same. I'm OK with changing the details of "-3 days 08:00". If somebody who used that in a plot config file now just gets "-3 days, current wall clock time" that's close enough.

kasemir, 2017-11-16:
Not saying we need to take the databrowser code hook, line and sinker.
Should certainly move to "Instant" etc.
I'm aware of the Duration and Period issue. In a way, they're correct because  'day' vs. '24 hours' is exactly where you need to treat it differently. Maybe that means our `RelativeTime` has internally both `Duration` and `Period`, with exactly one of them being `null` depending on the request being for "... hours, minutes, seconds" or "... days, weeks, month, years"?

Maybe we start with unit tests to describe what we need to see in the end, i.e. really TDT, and then we start to develop?
 * Parse "2017/01/17 13:45", "now" into instant
 * Parse "-3 days" or "3 days ago" into some relative time
 * Apply that relative time to 'noon' on a normal day, and one that's close to a DST change, to check if we again get 'noon' or are +-1 hour off
 * .. and so on

willrogers, 2017-11-16:
Perfect candidate for unit tests, agreed.

boryto, 2017-11-24:
Hi, 
for ITER  we also have to be able to parse ISO absolute time-date representations, the date formatter patterns:
DateTimeFormatter.ofPattern("yyyy-MM-dd'T'HH:mm:ss.nnnnnnnnnX"),
DateTimeFormatter.ofPattern("yyyy-MM-dd'T'HH:mm:ss.SSSX"),
DateTimeFormatter.ofPattern("yyyy-MM-dd'T'HH:mm:ssX"),
DateTimeFormatter.ofPattern("yyyy-MM-dd'T'HH:mmX"),
DateTimeFormatter.ofPattern("yyyy-MM-dd'T'HHX")

so the entered time for ex: 2017-11-20T14:40:15.123Z would be correctly parsed.

kasemir, 2018-02-01:
In core-util, org.phoebus.util.time has common formatters & parsers for absolute time, for relative times, and a combined `TimeRelativeInterval`.

In core-ui, org.phoebus.ui.time has the UI that shows all options in one pane.

In this example, both start and end time are relative:

[image]

Selecting an absolute start time changes the highlight so user knows it's now absolute start, relative end:
[image]

Example where both start and end are absolute, with calendar indicating the covered dates:
[image]

==> Done

closed 2018-02-01 by kasemir