ControlSystemStudio/phoebus #77: dependencies - target - parent

shroffk, 2017-11-03:
The dependency module and the parent could possibly be merged to reduce the number of build/target related modules.

We could create a single base module which
- serves as the main maven parent for all the phoebus modules
- produces the target which at first this will include both the lib and the maven repo...in the future once we have ensured that the ant builds are able to recursively search for jars in folder based repo structure we will have a single target repo.

questions:
should the merged project be located under 

phoebus/base/
ant_settings.xml
pom.xml
target/...

or should this be at the top level
phoebus
ant_settings.xml
pom.xml
target/...

I am leaving towards the second option

@kasemir @willrogers @claudio-rosati

kasemir, 2017-11-03:
I think that the 'top' module (i.e. combined dependencies + parent) should be a parallel directory to the rest and not a root directory because eventually you want to allow checking out only some parts, not all.

So actually, 'core' and 'applications' would also need to de-tangle.

In this example it's easy to have separate git repos for each of the folders:
```
top
core
core_framework
core_ui
applications
applications_probe
applications_display
applications_pvtable
product
```

top, core, applications can still provide some higher-level pom.xml and build.xml that's included with relative paths "../core/...." etc.

It's easy to only check out a sub-set and built:
```
top
core
core_framework
core_ui
applications
applications_pvtable
product
```

In contrast, with a directory tree like this, it's harder:
```
top
top/core
top/core/framework
top/core/ui
top/applications
top/applications/probe
top/applications/display
top/applications/pvtable
top/product
```

Would that be one "top" git repo? If not, how to you check out the separate repos into such a tree?
You'd have ".git' folders everywhere, so "git commit" on "top" would want to add all the subdirs.

shroffk, 2017-11-03:
ok, I agree...the "top" modules in parallel makes sense.

regarding the rest of the restructuring...I think the trick is going to be how to have the right number of modules.

I will start with the "top" for the time being.

closed 2022-08-10 by shroffk