ControlSystemStudio/phoebus #79: URI, URL, command line, input inconsistencies

kasemir, 2017-11-03:
The DockItemWithInput uses a URL as its input, for example
 * `file:/path/to/pv_table.pvs`
 * `http://server/path/to/display.bob?Macro1=Hello+Dolly&M2=abc`

There are valid URLs, they're stored in the memento with proper XML encoding.

The rest of the code often uses a URI (no 'schema:'), for example for opening tools from the command line:
 * `-app probe?pv=Fred`
 * `-app pv_table?file=file:/path/to/pv_table.pvs`

The `AppResourceDescriptor.create(String resource)` doesn't exactly say what type of resource text it expects. Could be a channel finder query, but most existing code now calls `ResourceParser.parseQueryArgs` in there and thus expects a command line type URI. Can't pass just "pv=Fred", must pass "probe?pv=Fred". So `PhoebusApplication.openResource` constructs that as `application.getName() + "?" + ResourceParser.FILE_ARG + "=" + resource`.

With the Display Builder runtime #78 there's now the issue that each instance is uniquely identified by the display file (local file or http://site/some_display.bob) combined with macros.
On the command line that would be
 * `-app display_runtime?file=http://server/path/to/display.bob&Macro1=Hello+Dolly&M2=abc`
In the DockItemWithInput, it turns into the input
 * `http://server/path/to/display.bob?Macro1=Hello+Dolly&M2=abc`

Note how one has "&Macro1" while the other has "?Macro1", so converting between the two formats, as right now would be required to re-open that display with same macros from the memento, isn't trivial.

Simply using
```
application.getName() + "?" + ResourceParser.FILE_ARG + "=" + resource
```
to turn a resource path into a URI is naive, since the resource could itself be "https://site/rest_api/displays/get?device=magnet&instance=2", so need to use
```
application.getName() + "?" + ResourceParser.FILE_ARG + "=" + URLEncoder.encode(resource, "UTF-8")
```

==> Turning into a mess, need a few better ideas

kasemir, 2017-11-03:
Idea 1: All one URI.

We force the command line argument to be a URI:
```
phoebus -app display_runtime?file=http://server/path/to/display.bob&Macro1=Hello+Dolly&M2=abc
```
So even a channel finder based tool will need to be called with a URL:
```
phoebus -app channel_list?pattern=..must_be_URLEncoded..
```

From then on, that URI is passed on without changes.
`AppResourceDescriptor.create(URI resource)` receives that URI, `DockItemWithInput`'s input uses that URI. Memento stores that URI.
The File/Open menu code converts the file name into a URI when opening the corresponding application.

kasemir, 2017-11-04:
Idea 2: Separate the application name from the resource URI that this application uses.

On the command line, one can either provide an application name
```
phoebus -app probe
phoebus -app display_editor
```
or just a resource URL (will be opened with suitable application. might ask 'which of these apps do you want?')
```
phoebus -resource file:/path/to/table.pvs
phoebus -resource file:/path/to/display.opi
```
or an application to use, followed by the resource to send to that application
```
phoebus -app probe -resource pv:?Fred
phoebus -app pv_table -resource file:/path/to/table.pvs
phoebus -app pv_table -resource http:/server.site/tables/table.pvs
phoebus -app pv_table -resource pv:?Fred&AnotherPV&YetAnotherPV
phoebus -app display_runtime -resource file:/path/to/display.opi
```

The resource must always be a URI (or even URL, as the above examples are all valid URLs?).
It will be passed on without changes.
`AppResourceDescriptor.create(URI resource)` receives that URI, `DockItemWithInput`'s input uses that URI. Memento stores that URI.

Advantage:
The resource URI (or URL) is passed without changes.
In most cases, except for many adding "file:" for a file, it's the natural file path or http:// URL that you'd have anyway for the resource.
A display editor and a display runtime that open the same resource would use the same text (in contrast to idea 1).

Disadvantage:
Need a URI/URL format for one or more PVs.

pavel-ch, 2017-11-07:
While thinking about these formats please consider also Windows users.
Just now, trying to open bob file in phoebus on Windows (yes it was possible to compile and start!) I have got:
```
Exception in thread "JavaFX Application Thread" java.lang.IllegalArgumentException: Illegal character in opaque part at index 2: C:\DATA\Epics\phoebus\applications\display\model\src\main\resources\examples\01_main.bob
	at java.base/java.net.URI.create(URI.java:881)
	at org.phoebus.framework.util.ResourceParser.createResourceURL(ResourceParser.java:42)
	........
Caused by: java.net.URISyntaxException: Illegal character in opaque part at index 2: C:\DATA\Epics\phoebus\applications\display\model\src\main\resources\examples\01_main.bob
	at java.base/java.net.URI$Parser.fail(URI.java:2912)
	.......
```
On Linux it works fine for me.

shroffk, 2017-11-07:
> ==> Turning into a mess, need a few better ideas

yes, these are becoming too many use case combinations with slight differences.

I like Idea 2, 

> pvs:?Fred&AnotherPV&YetAnotherPV

I think this is fine, the important thing is that all applications should use the same convention for defining the resource 

The initial goal with was to try to have a single URI encapsulate enough information so as to effectively launch or restore an application with its associated resource...

Since we are throwing around ideas, how about moving the app into the URI. A special keyword is used in the query section 
```
phoebus file:/path/to/table.pvs
phoebus file:/path/to/table.pvs?app=pv_table
phoebus file=http://server/path/to/display.bob&Macro1=Hello+Dolly&M2=abc&app=display_runtime
```

>	at org.phoebus.framework.util.ResourceParser.createResourceURL(ResourceParser.java:42)

I can look into this...maybe we can add tests to ensure that the ResourceParser works on all os's

kasemir, 2017-11-07:
Yes, that's the known issue #55

kasemir, 2017-11-07:
@shroffk Good, we both seem to like option 2:
 * Those apps that need a resource like "file://" or "http://.." can just receive that resource, we're not trying to wrap that into our own type of URI like "the_app?file=that_original_resource".

I assume it would be OK if we simply add the application name to the query of an existing resources, to

```
phoebus file:/path/to/table.pvs
```
opens that resource with the one and only application that handles *.pvs resources.
```
phoebus file:/path/to/table.pvs?app=pv_table
```
makes that explicit.
```
phoebus http://server/path/to/display.bob?Macro1=Hello+Dolly&M2=abc
```
might ask the user: Do you want to open with Display Editor or Display Runtime?
```
phoebus http://server/path/to/display.bob?Macro1=Hello+Dolly&M2=abc&app=display_runtime
```
avoids the prompt because the application name is provided.

Adding something like "app=" to the query should be OK if applications will simply ignore extra query parameters.
Only downside: You cannot use "app" as a macro any longer. So maybe we use "_app_", something that's less likely to be used as a "normal"?

Or go back to not changing the resource at all, the original option 2, so then the application name would need to come from a separate command line argument.

shroffk, 2017-11-07:
> Only downside: You cannot use "app" as a macro any longer.

I think this is a reasonable limitation.

Compared to the original option2 
- there is slightly less typing involved 
- we don't depend on users properly ordering the arguments

```
phoebus http://server/path/to/display.bob?Macro1=Hello+Dolly&M2=abc&app=display_runtime

phoebus -app display_runtime -resource http://server/path/to/display.bob?Macro1=Hello+Dolly&M2=abc
```
It also seems to be slightly better when dealing with a many files, i.e. launching 3-4 different bob files

```
phoebus 
http://server/path/to/display1.bob?Macro1=Hello+Dolly&M2=abc&app=display_runtime 
http://server/path/to/display2.bob?Macro1=Hello+Dolly&M2=abc&app=display_runtime 
http://server/path/to/display3.bob?Macro1=Hello+Dolly&M2=abc&app=display_runtime

phoebus
 -app display_runtime -resource http://server/path/to/display1.bob?Macro1=Hello+Dolly&M2=abc
 -app display_runtime -resource http://server/path/to/display2.bob?Macro1=Hello+Dolly&M2=abc
 -app display_runtime -resource http://server/path/to/display3.bob?Macro1=Hello+Dolly&M2=abc
```

I am fine with Option2 or modified Option2

kasemir, 2017-11-07:
Yes, in practice "app=.." should be OK. Certainly for macros, since those tend to be uppercase.

PR #85 (2017-11-16): Resource uris 79
This fixes #79 :

The 'input' to a DockItemWithInput is now a URI.
It's the same URI that can be passed via the "-resource /some/file.opi?MACRO=value" command line.
The "app=.." is treated as a special query parameter in case the resource needs to be opened with a specific application: "-resource /some/file.opi?app=some_app&MACRO=value".

The memento saves and restores that URI without changes, and the save/restore of display files with macros now works.

Documentation needs to be updated, and as I work further on the display builder there may be more tweaks, but this snapshot [... cut]
closed 2017-11-16 by shroffk