ControlSystemStudio/phoebus #80: org.w3c.dom resolution

shroffk, 2017-11-11:
While trying to setup the development environment with eclipse 4.7.1 and jdk9, there is some trouble with the core-util, pvtable, and display-runtime modules being unable to resolve the dependency to org.w3c.dom because it is available via java.xml and jython.standalone.

@kasemir Is there something I am missing in my setup?

kasemir, 2017-11-13:
I assume it builds with mvn as well as ant, only the Eclipse IDE notices the duplicate org.w3c.dom?

Cannot reproduce, yet, but looks similar to the  javax.annotation duplication in the email support, #52.
The only w3c class (interface) in python standalone is org.w3c.dom.html.HTMLDOMImplementation.
Will see if there's a jython jar other than 'standalone' that avoids the problem.
If not, since we use our own target platform, dependencies/target/lib, we could add a build step to the dependencies which strips w3c from the jython jar.

shroffk, 2017-11-13:
yes...
Both the builds work reliably fine (feels quite good saying that)

> If not, since we use our own target platform, dependencies/target/lib, we could add a build step to the dependencies which strips w3c from the jython jar.

Not a fan of this, we wanted to move away from eclipse, now we are repackaging stuff only for the eclipse IDE issues.

kasemir, 2017-11-13:
#81 proves that the mvn build is indeed fine.

Progress of sorts with Eclipse. I tried it with a new Eclipse setup:

Check Eclipse Preferences:
 - Java, Installed JREs: JDK 9 should be the default
 - Java, Compiler: JDK Compliance should be "9"

Use File, Import, General, Existing Projects into Workspace. Select the phoebus root directory, and check the option to "Seach for nested projects" to import _all_ projects

--> Can reproduce the numerous errors that are all based on duplicate for org.w3c.dom.html.HTMLDOMImplementation.

_Deleted_ the 'applications' and 'core' projects from the workspace, i.e. those 'parent' projects. Of course kept core-framework, core-*, applications-*, just deleted the top-level core and apps.

--> core-vtype misses src/main/test, which is mentioned in the .classpath but doesn't exist.
To fix, I created that 'test' dir, leaving it empty.

===> Now it works!!

Stopped Eclipse, deleted the workspace, started Eclipse again, this time importing everything

----> Get only the missing core-vtype/src/main/test error. Don't see the duplicate w3c problem, even though this time I have imported the parent core and applications projects?!

So the Eclipse error about the duplicate w3c may be valid, but I don't always get the error.
Can you also avoid the error by deleting core & apps from the workspace?

==> Then maybe we move on with that, because as you noted patching the released jython.jar isn't the best option, either.

kasemir, 2017-11-13:
Scratch that 'core', 'applications' idea.

The w3c error keeps resurfacing on my Mac OS setup.
On my Linux setup, when I use Navigate, Open Type to view HTMLDOMImplementation, Eclipse happily shows me that there are two, but it doesn't bother it for building.

shroffk, 2017-11-13:
> --> core-vtype misses src/main/test, which is mentioned in the .classpath but doesn't exist.
To fix, I created that 'test' dir, leaving it empty.

I was trying to commit a fix for this when I found the above error :)

kasemir, 2017-11-13:
One hint may be the exact error message: 
"The package org.w3c.dom is accessible from more than one module: java.xml, jython.standalone"

We're not using python.standalone as a _module_.

In 'dependencies', jython seems to end up on the module instead of the class path.
Checking how that happened.

kasemir, 2017-11-13:
Yes, that's it:
```
diff --git a/dependencies/.classpath b/dependencies/.classpath
index 91129c2..94fdc9f 100644
--- a/dependencies/.classpath
+++ b/dependencies/.classpath
@@ -26,10 +26,6 @@
        <classpathentry exported="true" kind="lib" path="target/lib/vtype-3.0.0.jar"/>
        <classpathentry exported="true" kind="lib" path="target/lib/javax.mail-1.6.0.jar"/>
        <classpathentry exported="true" kind="lib" path="target/lib/objenesis-1.0.jar"/>
-       <classpathentry exported="true" kind="lib" path="target/lib/jython-standalone-2.7.1.jar">
-               <attributes>
-                       <attribute name="module" value="true"/>
-               </attributes>
-       </classpathentry>
+       <classpathentry exported="true" kind="lib" path="target/lib/jython-standalone-2.7.1.jar"/>
        <classpathentry kind="output" path="bin"/>
```

kasemir, 2017-11-13:
Please try removing the "module" setting from dependencies/.classpath.
Combined with the core-vtype/src/main/test, I think that fixes it.

kasemir, 2017-11-13:
There still _is_ a problem in jython, which just like the email lib isn't ready to be used as a J9 module.
For for now we're not actually using modules, so should be fine until they're released as modules.

Jython was just updated in June, so somebody is still working on it.

shroffk, 2017-11-13:
Ok, this works...
Once I modified the .classpath the eclipse IDE errors are gone.

kasemir, 2017-11-13:
You wanna commit with some fix for the vtype test, either remove from .classpath or add at least one test to the dir?

shroffk, 2017-11-13:
I will do that...
I am cleaning up some other errors too

kasemir, 2017-11-13:
OK, I'll start on an update of the URIs, where we'll keep the plain URI for resources, not touching/rewriting it, but we'll recognize "...&app=some_app_name" as a hint which app to use, in case there's more than one.

shroffk, 2017-11-13:
should this be closed?

kasemir, 2017-11-13:
Yes, since we're both fine.
There's still the core-vtype/src/main/test, but I'll leave it to you if you want to remove test/ from the .classpath, or start adding tests to the dir so it's no longer empty and removed by git.

closed 2017-11-13 by kasemir