ControlSystemStudio/phoebus #86: Unable to build the project

claudio-rosati, 2017-11-15:
Hello,

I'm entered into the wacky world of Java 9, Pheobus and NetBeans 9... and I'm unable to build Phoebus.

I was able to open the maven project, and when I Clean & Build it this happens:

```
-------------------------------------------------------
 T E S T S
-------------------------------------------------------
Running org.phoebus.util.array.ListLongTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.067 s - in org.phoebus.util.array.ListLongTest
Running org.phoebus.util.array.ListFloatTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.004 s - in org.phoebus.util.array.ListFloatTest
Running org.phoebus.util.array.CollectionTest
Tests run: 6, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.017 s - in org.phoebus.util.array.CollectionTest
Running org.phoebus.util.array.ArrayFloatTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.014 s - in org.phoebus.util.array.ArrayFloatTest
Running org.phoebus.util.array.ListNumbersTest
Tests run: 33, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.067 s - in org.phoebus.util.array.ListNumbersTest
Running org.phoebus.util.array.ArrayBooleanTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.phoebus.util.array.ArrayBooleanTest
Running org.phoebus.util.array.CollectionNumbersTest
Tests run: 38, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.013 s - in org.phoebus.util.array.CollectionNumbersTest
Running org.phoebus.util.array.ArrayLongTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.phoebus.util.array.ArrayLongTest
Running org.phoebus.util.array.IteratorTest
Tests run: 6, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.011 s - in org.phoebus.util.array.IteratorTest
Running org.phoebus.util.array.ListIntTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 s - in org.phoebus.util.array.ListIntTest
Running org.phoebus.util.array.ListTest
Tests run: 5, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.007 s - in org.phoebus.util.array.ListTest
Running org.phoebus.util.array.BufferIntTest
Tests run: 5, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.phoebus.util.array.BufferIntTest
Running org.phoebus.util.array.ListByteTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 s - in org.phoebus.util.array.ListByteTest
Running org.phoebus.util.array.ListShortTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 s - in org.phoebus.util.array.ListShortTest
Running org.phoebus.util.array.ListMathTest
Tests run: 6, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.phoebus.util.array.ListMathTest
Running org.phoebus.util.array.ArrayShortTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 s - in org.phoebus.util.array.ArrayShortTest
Running org.phoebus.util.array.ListDoubleTest
Tests run: 9, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.006 s - in org.phoebus.util.array.ListDoubleTest
Running org.phoebus.util.array.ArrayDoubleTest
Tests run: 8, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 s - in org.phoebus.util.array.ArrayDoubleTest
Running org.phoebus.util.array.ArrayIntTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.phoebus.util.array.ArrayIntTest
Running org.phoebus.util.array.ArrayByteTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 s - in org.phoebus.util.array.ArrayByteTest
Running org.phoebus.util.array.CircularBufferDoubleTest
Tests run: 7, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.phoebus.util.array.CircularBufferDoubleTest
Running org.phoebus.util.time.TimeDurationTest
Tests run: 64, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.037 s - in org.phoebus.util.time.TimeDurationTest
Running org.phoebus.util.time.TimestampFormatTest
Tests run: 6, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.057 s - in org.phoebus.util.time.TimestampFormatTest
Running org.phoebus.util.time.TimeRelativeIntervalTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.phoebus.util.time.TimeRelativeIntervalTest
Running org.phoebus.util.time.TimestampTest
Tests run: 19, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 s - in org.phoebus.util.time.TimestampTest
Running org.phoebus.util.time.TimeParserTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.022 s - in org.phoebus.util.time.TimeParserTest
Running org.phoebus.util.time.TimeIntervalTest
Tests run: 12, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 s - in org.phoebus.util.time.TimeIntervalTest
Running org.phoebus.util.text.NumberFormatsTest
Tests run: 3, Failures: 2, Errors: 0, Skipped: 0, Time elapsed: 0.014 s <<< FAILURE! - in org.phoebus.util.text.NumberFormatsTest
format1(org.phoebus.util.text.NumberFormatsTest)  Time elapsed: 0.012 s  <<< FAILURE!
java.lang.AssertionError: 

Expected: "2.00"
[... 77 lines cut ...]
	at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Launcher.java:289)
	at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.java:229)
	at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(Launcher.java:415)
	at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:356)
Caused by: org.apache.maven.plugin.MojoFailureException: There are test failures.

Please refer to /Users/claudiorosati/Projects/GitHub4/phoebus/core/util/target/surefire-reports for the individual test results.
Please refer to dump files (if any exist) [date]-jvmRun[N].dump, [date].dumpstream and [date]-jvmRun[N].dumpstream.
	at org.apache.maven.plugin.surefire.SurefireHelper.throwException(SurefireHelper.java:204)
	at org.apache.maven.plugin.surefire.SurefireHelper.reportExecution(SurefireHelper.java:98)
	at org.apache.maven.plugin.surefire.SurefirePlugin.handleSummary(SurefirePlugin.java:334)
	at org.apache.maven.plugin.surefire.AbstractSurefireMojo.executeAfterPreconditionsChecked(AbstractSurefireMojo.java:937)
	at org.apache.maven.plugin.surefire.AbstractSurefireMojo.execute(AbstractSurefireMojo.java:785)
	at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo(DefaultBuildPluginManager.java:134)
	at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:207)
	... 20 more

Re-run Maven using the -X switch to enable full debug logging.

For more information about the errors and possible solutions, please read the following articles:
[Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException

After correcting the problems, you can resume the build with the command
  mvn <goals> -rf :util
```

kasemir, 2017-11-15:
The error seems related to locale, the formatting of numbers:
```
Expected: "2.00"
     but: was "2,00"
	at org.phoebus.util.text.NumberFormatsTest.format1(NumberFormatsTest.java:26)
```

Looks like nobody ran that test in Europe before.

There have been discussions on how we format number, timestamp, ...

I'm of the conviction that this is control system code. It's used in control rooms or on beam lines where you often have visitors from around the world. English tends to be the only common denominator.
Plus we sometimes export data files.
==> Numbers should always be represented as "3.14", time stamps as "2017-01-18 13:45:26", so no matter who executes the code where, you get the same result.

The other opinion is that formats should be based on locale, plus allow arbitrary customization
==> Numbers could be "3,14", time stamps could be "18ter Tag im Januar des Jahres 2017"

kasemir, 2017-11-15:
You could try to modify the locate settings to EN/US before running the test, https://stackoverflow.com/questions/8809098/how-do-i-set-the-default-locale-for-my-jvm

willrogers, 2017-11-15:
I tend to agree with Kay - we choose an ISO 8601 time and date format and we stick to it.

claudio-rosati, 2017-11-15:
What about putting it as default into a property file?

kasemir, 2017-11-15:
For now, if you want the test to pass, update the NumberFormatsTest constructor to look like this:
```
    public NumberFormatsTest() {
        Locale.setDefault(Locale.US);
    }
```

kasemir, 2017-11-16:
Try again with the latest from master which includes the Locale setup for the test

claudio-rosati, 2017-11-17:
@kasemir @shroffk 
I've fixed `TemporalRoundingTest` because it has the same Locale issue.
I was sure I worked on a forked version, so I committed the change for a following pull request. Only at that point I've discovered I worked on a cloned copy, not a fork. I apologize for that.
Now I have cloned a forked version of Phoebus, so kind of mistakes will be avoided.

claudio-rosati, 2017-11-17:
The main branch still cannot be built because of the following:

```
Expected: "'1E0' 2E0 3E0 4E0 5E0 6E0 7E0 8E0 9E0 '1E1' 1E1 2E1 3E1 4E1 5E1 6E1 7E1 8E1 9E1 '1E2' 1E2 2E2 3E2 4E2 5E2 6E2 7E2 8E2 9E2 '1E3' 1E3 2E3 3E3 4E3 5E3 6E3 7E3 8E3 9E3 '1E4' "
     but: was "'1×10^0' 1×10^1 2×10^1 3×10^1 4×10^1 5×10^1 6×10^1 7×10^1 8×10^1 9×10^1 '1×10^2' 1×10^3 2×10^3 3×10^3 4×10^3 5×10^3 6×10^3 7×10^3 8×10^3 9×10^3 '1×10^4' "
java.lang.AssertionError
Expected: "'1E0' 2E0 3E0 4E0 5E0 6E0 7E0 8E0 9E0 '1E1' 1E1 2E1 3E1 4E1 5E1 6E1 7E1 8E1 9E1 '1E2' 1E2 2E2 3E2 4E2 5E2 6E2 7E2 8E2 9E2 '1E3' 1E3 2E3 3E3 4E3 5E3 6E3 7E3 8E3 9E3 '1E4' "
     but: was "'1×10^0' 1×10^1 2×10^1 3×10^1 4×10^1 5×10^1 6×10^1 7×10^1 8×10^1 9×10^1 '1×10^2' 1×10^3 2×10^3 3×10^3 4×10^3 5×10^3 6×10^3 7×10^3 8×10^3 9×10^3 '1×10^4' "
	at org.csstudio.javafx.rtplot.LogTicksTest.testLogTicks(LogTicksTest.java:35)
```

kasemir, 2017-11-17:
You get "1×10^1"? That's cool!
What is your locale?
```
        System.out.println(Locale.getDefault());
```

claudio-rosati, 2017-11-17:
CI0011766:/ claudiorosati$ jshell
|  Welcome to JShell -- Version 9.0.1
|  For an introduction type: /help intro

jshell> System.out.println(Locale.getDefault());
**en_SE**

kasemir, 2017-11-17:
Ah, good. Didn't see that in the Locale constants, but makes sense for Sweden.

This commit fixed it: b978142eba4b4652871b36118d1b9371b05e7278

claudio-rosati, 2017-11-17:
OK, I'll test it when in master branch.

kasemir, 2017-11-17:
That commit is in the display runtime branch where I'm currently working.

claudio-rosati, 2017-11-17:
```
Running org.csstudio.javafx.rtplot.LogTicksTest
Ticks for 1.0 .. 10000.0:
'1×10^0' 1×10^1 2×10^1 3×10^1 4×10^1 5×10^1 6×10^1 7×10^1 8×10^1 9×10^1 '1×10^2' 1×10^3 2×10^3 3×10^3 4×10^3 5×10^3 6×10^3 7×10^3 8×10^3 9×10^3 '1×10^4' 
Tests run: 3, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 2.372 s <<< FAILURE! - in org.csstudio.javafx.rtplot.LogTicksTest
testLogTicks(org.csstudio.javafx.rtplot.LogTicksTest)  Time elapsed: 1.621 s  <<< FAILURE!
java.lang.AssertionError: 

Expected: "'1E0' 2E0 3E0 4E0 5E0 6E0 7E0 8E0 9E0 '1E1' 1E1 2E1 3E1 4E1 5E1 6E1 7E1 8E1 9E1 '1E2' 1E2 2E2 3E2 4E2 5E2 6E2 7E2 8E2 9E2 '1E3' 1E3 2E3 3E3 4E3 5E3 6E3 7E3 8E3 9E3 '1E4' "
     but: was "'1×10^0' 1×10^1 2×10^1 3×10^1 4×10^1 5×10^1 6×10^1 7×10^1 8×10^1 9×10^1 '1×10^2' 1×10^3 2×10^3 3×10^3 4×10^3 5×10^3 6×10^3 7×10^3 8×10^3 9×10^3 '1×10^4' "
	at org.csstudio.javafx.rtplot.LogTicksTest.testLogTicks(LogTicksTest.java:35)
```

```
Tests run: 6, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 0.068 s <<< FAILURE! - in org.csstudio.javafx.rtplot.util.TemporalRoundingTest
testDST(org.csstudio.javafx.rtplot.util.TemporalRoundingTest)  Time elapsed: 0.039 s  <<< FAILURE!
java.lang.AssertionError: 

Expected: "2014-03-08 23:00:00 (EST, UTC -0500)"
     but: was "2014-03-08 23:00:00 (GMT-05:00, UTC -0500)"
	at org.csstudio.javafx.rtplot.util.TemporalRoundingTest.testDST(TemporalRoundingTest.java:191)
```

```Tests run: 4, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 0.006 s <<< FAILURE! - in org.csstudio.javafx.rtplot.LinearTicksTest
testNormalTicks(org.csstudio.javafx.rtplot.LinearTicksTest)  Time elapsed: 0.003 s  <<< FAILURE!
java.lang.AssertionError: 

Expected: "-0.6 -0.5 -0.5 -0.4 '-0.4' -0.4 -0.3 -0.3 -0.2 '-0.2' -0.2 -0.1 -0.1 -0.0 '0.0' 0.0 0.1 0.1 0.2 '0.2' 0.2 0.3 0.3 0.4 '0.4' 0.4 0.5 0.5 0.6 "
     but: was "-0,6 -0,5 -0,5 -0,4 '-0,4' -0,4 -0,3 -0,3 -0,2 '-0,2' -0,2 -0,1 -0,1 -0,0 '0,0' 0,0 0,1 0,1 0,2 '0,2' 0,2 0,3 0,3 0,4 '0,4' 0,4 0,5 0,5 0,6 "
	at org.csstudio.javafx.rtplot.LinearTicksTest.testNormalTicks(LinearTicksTest.java:66)
```

kasemir, 2017-11-17:
Right, you need to add the `Locale.setDefault(Locale.US);` from b978142. You could cherry pick that commit, or wait until I have a merge request for that branch.

kasemir, 2017-11-30:
Looks like all these build errors were really test failures, so `mvn -DskipTests clean install` would have worked.

Still, the locale-based tests have been updated.

I just built the latest (display_editor2 branch):
````
$ export MAVEN_OPTS="-Duser.country=SE -Duser.language=en"
$ mvn --version
...
Java version: 9.0.1, vendor: Oracle Corporation
...
Default locale: en_SE, platform encoding: UTF-8
...
$  mvn clean install 
...
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 02:01 min

$ ant clean dist
...
dist:
      [zip] Building zip: /home/ky9/git/phoebus/phoebus-product/target/phoebus-0.0.1.zip

BUILD SUCCESSFUL
Total time: 12 seconds
```

Can we close this?

closed 2018-03-28 by kasemir