ControlSystemStudio/phoebus #93: Display Builder for Phoebus - site specific widgets

pavel-ch, 2017-12-05:
Is there any plan to allow using site specific Display Builder widgets / widget libraries?
- allow a site / user to implement a specific functionality into a widget and use it transparently in BOB screens
- allow an easy deployment scheme (plain copy a Java 9 module into a specific Phoebus installation directory?)
This demand arises for sure earlirer or later as soon as Phoebus and Display Builder gets a wider user base.
Just talking about plans, not about an immediate implementation.

kasemir, 2017-12-05:
Basically the same as it was done via RCP extension points for the Eclipse-based Display Builder,
widgets (model, representation, runtime) are provided via SPI:

 * https://github.com/shroffk/phoebus/blob/master/applications/display/model/src/main/java/org/csstudio/display/builder/model/spi/WidgetsService.java
 * https://github.com/shroffk/phoebus/blob/master/applications/display/representation/src/main/java/org/csstudio/display/builder/representation/spi/WidgetRepresentationsService.java
 * https://github.com/shroffk/phoebus/blob/master/applications/display/runtime/src/main/java/org/csstudio/display/builder/runtime/spi/WidgetRuntimesService.java

See for example how https://github.com/shroffk/phoebus/blob/master/applications/display/model/src/main/resources/META-INF/services/org.csstudio.display.builder.model.spi.WidgetsService registers the base widget models.

So all you'd have to do is build a product similar to the generic  phoebus-product, but including your site-specific modules in your variant of https://github.com/shroffk/phoebus/blob/master/phoebus-product/pom.xml

pavel-ch, 2017-12-06:
Thank you. I give it a try, closing the issue and eventually reopening for questions.

closed 2017-12-06 by pavel-ch

kasemir, 2017-12-17:
As for creating a site-specific product, https://github.com/kasemir/phoebus-sns is the start of an SNS-specific product.
It uses SPI to add PV name completion.
Similarly, such a site-specific product could widgets via SPI.