ControlSystemStudio/phoebus #108: Java 9 and JavaFX

willrogers, 2018-01-19:
I downloaded Java 9 from the location in README.md (http://jdk.java.net/9/) and I didn't get JavaFX bundled with my download.

When I downloaded it from Oracle (http://www.oracle.com/technetwork/java/javase/downloads/jdk9-downloads-3848520.html) I got JavaFX.

Do we still need Oracle JDK to use JavaFX?

berryma4, 2018-01-19:
> technetwork/java/javase/downloads/jdk9-downloads-3848520.html) I got

openjfx is a separate package:
http://openjdk.java.net/projects/openjfx/

willrogers, 2018-01-19:
I went to that page, and didn't find a download.

Is there another way to download the jfx modules?

kasemir, 2018-01-19:
Right, no luck here, either.
The OpenJFX wiki page leads to "Community Builds", https://wiki.openjdk.java.net/display/OpenJFX/Platforms, which is empty/obsolete.

berryma4, 2018-01-19:
It's in debian and ubuntu:
https://packages.ubuntu.com/xenial/openjfx

willrogers, 2018-01-19:
I can't find much for RHEL. Red Hat suddenly pulled support for Oracle Java 8 last year so I would rather use OpenJDK 9 if we can.

berryma4, 2018-01-19:
Although, personally, I'm using oracle for linux and windows.

berryma4, 2018-01-19:
Not much action here:
https://answers.launchpad.net/ubuntu/+source/openjdk-9/+question/658982

But it looks like debian is working on it as we speak:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=850921

kasemir, 2018-03-27:
Meanwhile, downloaded the recently released Oracle Java 10.
Phoebus seems to run fine with it.

berryma4, 2018-03-28:
Also, for JDK 11, javafx will be a separate download (like openjfx)
https://blogs.oracle.com/java-platform-group/the-future-of-javafx-and-other-java-client-roadmap-updates

berryma4, 2018-03-28:
we can probably close this ticket now.

closed 2018-03-28 by berryma4