ControlSystemStudio/phoebus #112: Embedded Display Widget - Transparent Group handling

pavel-ch, 2018-01-27:
As suggested some time ago in [https://github.com/ControlSystemStudio/cs-studio/issues/2055](url), transparency of Embedded Widget can greatly help with displays construction using functional decomposition. This issue offers an automatic transparency of EmbeddedDisplay widget when hosting a transparent Group, preserving all other features and use cases. Pull request follows.

kasemir, 2018-01-29:
I think the idea of a transparent background for the embedded widget is generally useful.
I'll add that as a plain "transparent" option, which is easier to understand than automagically getting a transparent background but only when using the a subgroup of an embedded display.

pavel-ch, 2018-01-30:
Ok for me. Only 1 case can be confusing: showing transparent Group in a non-transparent Embedded widget shows the (in the template display invisible) Group background color. But this seems to be an edge case, because now the Group does not require to be transparent itself, so for me this is acceptable.

kasemir, 2018-01-30:
> .. transparent group in non-transparent embedded widget ..

I assume you're referring to the case where you embed not a complete display, but you select just one group of the embedded display via that 'group' option of the embedded display.

In that case, maybe the 'transparent' option of the embedded display widget should be set to the value of the 'transparent' option of the target group? After all, the content of the embedded display is then just that one group, and there really isn't any other background color for the embedded content beyond the color of the group.

pavel-ch, 2018-01-30:
> I assume you're referring to the case where you embed not a complete display, but you select just one group of the embedded display via that 'group' option of the embedded display.

Yes

> In that case, maybe the 'transparent' option of the embedded display widget should be set to the value of the 'transparent' option of the target group? After all, the content of the embedded display [...]

Good idea. It is a bit automagical but would work logically as expected.

kasemir, 2018-02-07:
Closing this, I think it's handled.

closed 2018-02-07 by kasemir

kasemir, 2018-03-08:
@pavel-ch  Pavel, the transparency handling had an unexpected side effect..

It used to simply set the ScrollPane's style to `-fx-background: null`, which caused two
problems:

1) It resulted in "cannot parse color" errors from the CSS layer

2) It was inherited by the embedded content, so Text Entry fields would
also get a transparent background (https://github.com/kasemir/org.csstudio.display.builder/issues/360)

In an update, I checked the ScrollPaneSkin and modena.css, found the ".viewport", and now generally make the scroll pane transparent.
The actual color or transparency of the container's background is then controlled via the "inner" pane of the EmbeddedWidgetRepresentation, which seems to work OK.

pavel-ch, 2018-03-08:
Seeit now. Thank you for solving the issue. I was not aware of these css side effects, but they are logical. I will try to test it on my displays tomorrow.