ControlSystemStudio/phoebus #115: Layout Switching

kasemir, 2018-02-01:
Phoebus supports saving the current layout (window positions, tabs in each window, memento detail of each tab) on shutdown and re-loading it on startup.

That should be expanded to support saving and restoring named layouts at runtime:
 * Preference that defines a directory used for layout files.
 * Menu option to save the current layout into a file
 * Menu that lists layout files and allows selecting one

Switching layouts will close the currently open windows and open the new ones.

pavel-ch, 2018-02-13:
I would recommend to add one more feature to layouts:
allow interactions from inside Display Builder actions to layout positions:
- action that opens opi/bob runtime at a (somehow addressed or named ) position in layout
- action that opens any other phoebus application file at a (somehow addressed or named ) position in layout

kasemir, 2018-06-06:
The discussion of additional options to control what opens where is on #148 and #174.
Basic save/load of layout as described in this ticket has been implemented.

closed 2018-06-06 by kasemir