ControlSystemStudio/phoebus #117: File/Open handling when there are multiple options

kasemir, 2018-02-09:
When File/Open is called for *.opi or *.bob files, which can be handled by both the display editor and runtime, the user currently gets a dialog to select the desired tool, which gets old.

Ideas:

**Default application**
Make one of the applications a 'default', in this example the display runtime.
So File/Open *.opi will always open the display runtime, from where one can use the context menu to open the display in the editor.

**File/Open With**
In _addition_ to the default handling, add menu File/Open With that will present the dialog for selecting a tool. So File/Open *.opi will get the runtime, File/Open With prompts for editor or runtime.

**File/Open With updates the default application**
.. as in RCP. After Open With was invoked to open in editor, File/Open will now default to editor.
For control room or beam line this is probably a bad idea, because what a user gets from File/Open depends on what the previous user did via File/Open With..

** Magically know what the user really wants**
Best, but unclear how to implement.

pavel-ch, 2018-02-13:
Opinion:
Default application - definitely yes. Allow setting.ini to set one (technician sets opi editor in his workspace/computer, operator workstation has the runtime set as default)
File/Open With - yes to allow to use a non-default app
File/Open With updates the default application - for me impractical, does not work well, I hate it in RCP
Magically know what the user really wants - no idea

kasemir, 2018-02-20:
I think we agree on all: Open With .. to give choice, plain open to use a default.
Yes, using the last "open with" to update the default as in RCP has been a bother to me as well.
I had originally thought that the default is mostly static. An application would declare itself the "default".
But I like your idea of making that a preference, so that's what it now is.
Originally set to use the "*run*" type application, but can be changed to "*edit*" for those who mostly create displays.

PR #125 (2018-02-22): File open , open with.., default app
Fixes #117 

Assume a *.bob file. Both display_runtime and display_editor can handle this resource.

With File/Open With, a prompt asks the user which app to use.

With plain File/Open, or "-resource abc.bob" from the command line, without "?app=..", a new preference setting now decides which app to use.
```
default_apps=run
```
Format is comma-separated list with sub-text (!) of application name.
For example, "run, exe" would pick "display_runtime" over "display_editor", and "foo_executor" over "foo_creator".
The patterns "edit, creat" would inversely open [... cut]
closed 2018-02-20 by kasemir