ControlSystemStudio/phoebus #120: Display builder - Zoom and Pan mouse wheel gestures for both Editor and Runtime

pavel-ch, 2018-02-18:
From my experience, it seems to be really useful to have these mouse gestures in OPI software:
- Zoom with Ctrl-mouse wheel
- Pan with mouse wheel/middle button press and drag

I tried to implement both. The implementation is a working prototype implementation. Still two known issues which I can solve as soon as I know you are interested in:
- Zoom: return to a previous "combo driven" zoom level from any wheel level is not possible directly (no value change in combo)
- Pan: panning above mouse events consuming widgets (Controls, Embedded ...) is not possible.

I hope you are interested in this feature, and I hope I selected the right place for the code.

Pull request #119

kasemir, 2018-02-19:
That's good, thanks!

kasemir, 2018-02-19:
Adjusted it slightly to be compatible with the RCP version:
The RCP runtime has an SWT combo box.
So instead of `setZoomAction(JFX Combo)` it's now a more generic `setZoomListener(String)` which can be used with a handler that updates a JFX combo or SWT combo with the received text.

pavel-ch, 2018-02-20:
To the event filtering/handling (Don't _filter_ because some widgets (plot) also handle the wheel):
- I did not see that plot is also handling this event. My mistake.
Need some reconsideration, because the new approach with addEventHandler() does not work well for me: it zooms BUT also scrolls at the same time. Evidently the scroller scrolls BEFORE our handler is called.
Also found a bug - the selection frames of editor are not zoomed together with content, trying to find solution.

pavel-ch, 2018-02-20:
#122 solves the above: 
found a bug - the selection frames of editor are not zoomed together with content, trying to find solution.

pavel-ch, 2018-02-20:
Some observations to addEventHandler():
- the behaviour now is: zoom only works when scrollbars are at the end position in the given direction.
We must perhaps find another solution. This Ctrl-wheel conflict with Databrowser plot can be fatal.
Still Googling.

kasemir, 2018-02-20:
Thanks for looking at the details, and please continue to tweak until it's correct.

Also keep in mind that this is for a control system, not for e.g. generic web pages.
The displays should in general be created such that they open OK on the designated control room computer. Operators should not have much need to zoom & pan. If they do, the screen was not the correct size/layout to begin with.

Inside a plot or image, however, the operator will sometimes need to zoom & pan.
So if there's any question about priorities, the wheel events need to be handled by the plot more than the display runtime.

In the editor, zoom & pan can be useful, typically zooming in to better align some widgets, then back to "100%".

In principle, that should all work out with `model_root.addHandler`, so you can zoom/pan the display as long as the mouse is on an empty space, otherwise the event goes to the widget. But the devil may be in the details, as you saw with scrollers capturing the event before our handler is called.

If all fails, we could use another key modifier, i.e. do use `model_root.addFilter`, but require for example Crtl & Shift. A bit inconvenient, since 'Ctrl + Wheel' is the common way to zoom...

kasemir, 2018-02-20:
Note that the months-long 'databrowser' work has been wrapped up, so right now it's all back on 'master'.

pavel-ch, 2018-02-21:
OK I can try to solve the issue. Using Master branch from now on.
To pan & zoom for operators: we use this feature for electric substations: having an overview scheme (several substations) you can zoom in to a partiicular scheme part (a substation or even one feeder). This is connected with another features:
- the possibility to react to the zoom change to for example show/hide widgets.
- the possibility to pan to a specified position or PV name occurence by an action (issued for example from another display)
This is next step of my investigation in long term. For now, pure zoom/pan is ok for me as soon as it works correctly.
May be useful for you , too?

pavel-ch, 2018-02-23:
What about this:
Editor: Widget internal wheel functions not needed -> can "filter" events
Runtime: a new Display Property, something like "Allow runtime Ctrl-wheel zoom"? Defaulting to false, and if false, effectively disables filter (does not capture events).

kasemir, 2018-02-23:
> Runtime: a new Display Property, something like "Allow runtime Ctrl-wheel zoom"?

Doesn't that mean you need to re-implement the JFX event bubbling?
You'd filter the event, then inside the filter locate the widget, ask it how it would like to handle the even, ..

Then I'd rather just use `addEvent` on the root, so to use the wheel zoom, the mouse needs to be outside a widget. If the mouse is on top of a widget, the event goes to that widget, which in the case of the plot would zoom/pan the plot. Other widgets don't do anything, which is fine. To scroll/zoom the whole display, need to be on some part of the widget's background.

pavel-ch, 2018-02-23:
To Editor: am I right - can use EventFilter?
To Runtime - 
`need to re-implement the JFX event bubbling`
Not that ambitious, I was thinking about a simple static user defined Display model boolean property for a display, not for all widgets - to sort out diplays with/without desired runtime wheel zoom function. But I will consider your suggestion.

pavel-ch, 2018-02-25:
Tried to resolve issues.
Split Edit and Runtime event handling. Using @kasemir suggestion for Runtime wheel zoom.
#127

kasemir, 2018-02-26:
Seems to work fine now in both editor and runtime. Thanks!

closed 2018-03-20 by kasemir