ControlSystemStudio/phoebus #126: Support opening standalone web browser from top_resources

kasemir, 2018-02-23:
Need to somehow support opening standalone web browser from top_resources.
This way, links to web pages can be added to the File/Top Resources.

Should end up calling `javafx.application.HostServices.showDocument()`, i.e. open the usual web browser, outside of the Java app.
Could be implemented directly in `PhoebusApplication.createTopResourcesMenu()`. Maybe there's also an advantage in implementing this via an `AppDescriptor`, in which case resources might require the form `http://path/to/page?app=that_new_app_which_starts_external_web_browser`.

shroffk, 2018-02-23:
I think implementing this as an app might be a good idea.

The app would contribute a menu item to the top menu...additionally we could have the app support all actions of opening webpages (e.g. a context action on a alarm)

kasemir, 2018-06-21:
Implemented app=web, which launches the normal web browser, with example top_resources entry

PR #195 (2018-06-22): Add "web" app that opens URLs in external web browser
.. and example how to use that from top resources menu.

Fixes #126
closed 2018-06-22 by kasemir