ControlSystemStudio/phoebus #135: File Browser

kasemir, 2018-03-19:
Desired functionality (still open to discussion) and progress on the file browser introduced in #134:

 * [x] Browse files
 * [x] Context menu to "Open" file in appropriate app
 * [x] Allow selecting app if there's more than one
 * [ ] Show error when file can't be opened in dialog (in addition to the current log message)
 * [x] Context menu to "Open With" to pick app
 * [ ] Double-click to "Open"
 * [ ] Allow opening in external tool
 * [ ] Icon of the (default) application
 * [ ] Allow opening file in editor (is that just one of the external tools?)
 * [ ] Save/restore state. AppResourceDescriptor with create(URL) so that it allows opening navigator for directory, keeping that as the file:// resource which is saved/restored on restart? Or use memento for the URL? Or use AppResourceDescriptor for URL, plus memento for the currently selected file, expanding file tree to that on restart?
 * [ ] Hide 'hidden' files based on `File.isHidden()`
 * [ ] Context menu: Rename files
 * [ ] Context menu: Copy/paste/duplicate files
 * [ ] Allow 'dropping' files from outside to copy or move them into the target directory
 * [ ] Context menu: Create new files?
 * [ ] Context menu: Show/edit properties (executable, read-only)?
 * [ ] Copy full path of file to clipboard (so can then use 'chmod' etc. in terminal)

kasemir, 2018-03-19:
Already works quite well, so far no performance issues when browsing files!

Small layout issue in the top row on Linux, where label and button are too narrow:
[image]

shroffk, 2018-03-19:
Ok...I can clean that up

The double click open is also implemented...it opens using the default app or prompts the user to select an app when multiple apps can handle the selected file.

kasemir, 2018-03-19:
Regarding double-click, I think it should be like File/Open: If there are multiple apps, it uses the preference to pick the default. (See use of `Preferences.default_apps` in `PhoebusApplication`.(

Right now double-click is like File/Open With:  If there are multiple apps, it prompts. You can already to that via the context menu, Open With. The double click should be the fastest possible operation, using the Preferences.default_apps, or the first one found, never prompting.

kasemir, 2018-06-29:
File browser works nicely when in a "Locked" (frozen) pane: Double-clicking on a file opens the tab in the next non-frozen pane.

[image]

kasemir, 2018-07-02:
Here's a layout in code as opposed to FXML what resizes properly without cutting the label and button in the top row:

```
Label label = new Label("Base Directory:");
TextField path = new TextField();
Button browse = new Button("Browse");

label.prefHeightProperty().bind(path.heightProperty());
HBox.setHgrow(path, Priority.ALWAYS);
HBox toolbar = new HBox(5.0, label, path, browse);

TreeView<File> tree = new TreeView<>();

BorderPane layout = new BorderPane(tree);
toolbar.setPadding(new Insets(0, 0, 5, 0));
layout.setTop(toolbar);
layout.setPadding(new Insets(5.0));
```

[image]

No idea how to express that in the FXML.

kasemir, 2018-07-02:
Never mind, figured it out.

kasemir, 2018-07-25:
Removing 'create new file', because we don't have a mechanism for creating the initial content.
For example, a new Data Browser *.plt is best created via Applications, Display, Data Browser and then saving it. Creating an empty *.plt file won't work.

All the other ideas listed have been implemented.

closed 2018-07-25 by kasemir