ControlSystemStudio/phoebus #143: Java 10, Java 11

kasemir, 2018-04-19:
Phoebus runs OK with Java 10.

The one new thing in Java 10 is 'var', but I don't see a desperate need to use that and break Java 9 compatibility.

Creating this ticket to track progress with Java 11:

 * Announcement that JDK 11 will no longer include JFX, https://blogs.oracle.com/java-platform-group/the-future-of-javafx-and-other-java-client-roadmap-updates

 * Oracle white paper "Java Client Roadmap" http://www.oracle.com/technetwork/java/javase/javaclientroadmapupdate2018mar-4414431.pdf

 * OpenJDK "making sure .. can easily create and run JavaFX applications [in] Java 11", 
https://github.com/javafxports/openjdk-jfx/issues/52

claudio-rosati, 2018-04-19:
I don’t think I’ll never use ‘var’: from a maintenance point of view is one of the worst thing.

kasemir, 2018-09-28:
Java 11 Update (http://mail.openjdk.java.net/pipermail/openjfx-dev/2018-September/022475.html):

 * New OpenJFX community site http://openjfx.io/
 * "final release of JavaFX 11" available as download from that site or maven central as openjfx:javafx-COMPONENT:11 with COMPONENT=base, graphics, controls, ...
 * Runs with preview of JDK 11 or JDK 10 (OpenJDK build only)

I take this to mean:
Once we switch to Java 11, we add  openjfx:javafx-*:11 to the dependencies, and expect that to work with both Oracle and OpenJDK 11, neither of which will bundle JFX.
Java 9 and 10 will then need to be abandoned, because Oracle9/10 include JFX and that would clash with the JFX11 from the dependency.

willrogers, 2018-09-28:
I have someone working on building and running CS-Studio with Java 11 - OpenJDK plus OpenJFX.

claudio-rosati, 2018-09-28:
Next week I'll start looking to extend the Integration plugin to open Display Builder and Databrowser 3 in Phoebus from CS-Studio.

shroffk, 2018-09-28:
> Oracle9/10 include JFX and that would clash with the JFX11 from the dependency.

I am having trouble with travis and openjdk + javafx

The trouble is that openjfx is no longer available for trusty. It is available for xenial and bionic distributions of ubuntu...but travis still does not have support for those distributions.

So when we move forward to 11, we might have to rewrite the travis build files.

shroffk, 2018-09-28:
> Next week I'll start looking to extend the Integration plugin to open Display Builder and Databrowser 3 in Phoebus from CS-Studio.

That is great, I think we need to split the current integration plugin, which was written as a proof of concept, into individual plugins - one for each application being integrated.

```
/phoebus/phoebus-plugins/org.csstudio.phoebus.integration.databrowser
/phoebus/phoebus-plugins/org.csstudio.phoebus.integration.probe
/phoebus/phoebus-plugins/org.csstudio.phoebus.integration.logbook
```
this way each product can be created by selectively including only those applications they wish to move to phoebus

claudio-rosati, 2018-09-28:
Yes, I agree. I'll try, starting from the current one, to create the two for Display Builder and Databrowser.

berryma4, 2018-10-04:
This is great news!! This seems much easier.
I wonder if we can "just" add them to maven-osgi-bundles for the RCP Eclipse/cs-studio @willrogers

kasemir, 2018-10-04:
The java11 branch still compiles and runs with Oracle JDK10, but not sure how that works.

Compiling with JDK10, it fetches the new JavaFX11 as configured in the pom.xml, and that's included in the product. But at runtime you see `javafx.version = 10.0.1`, so it's still using the JFX bundled into the Oracle JDK10.

When compiled with Oracle Java 11, at runtime you see that it's using the javafx jars bundled in the product, not the ones from the JDK (since there aren't any in the JDK):
```
java.vm.vendor = Oracle Corporation
java.vm.version = 11+28
javafx.runtime.version = 11+26
javafx.version = 11
```

The javafx jars include OS-specific parts. When compiled on Mac OS X, the Phoebus product includes these:
```
javafx-base-11-mac.jar
javafx-base-11.jar
javafx-controls-11-mac.jar
javafx-controls-11.jar
...
```

So will have to compile for each OS.

kasemir, 2018-10-05:
It's possible to build cross-platform like this:

```
mvn -Djavafx.platform=mac   -DskipTests clean install
mvn -Djavafx.platform=linux -DskipTests clean install
mvn -Djavafx.platform=win   -DskipTests clean install
```

The result of each built would have to be packaged/deployed before starting the build for the next platform, since that cleans the previous one.

kasemir, 2018-10-05:
Similar, for the `ant` build this approach means 3 different target platforms.
As long as a build takes 1-2 minutes, that might be just fine.

The alternative, I think, would require creating a target platform that has binaries for all `javafx.platform` options, and then adding the `javafx.platform` to the names of the product/lib etc.

kasemir, 2018-10-05:
@shroffk Do you know the state of the PVA client lib requiring java.corba?
That module is no longer available with OpenJDK11.

willrogers, 2018-10-08:
I don't know why PVA requires corba, but I managed to build CS-Studio against `org.jacorb.omgapi`.

willrogers, 2018-10-08:
@berryma4 I'm happy to try adding JavaFX as Maven dependencies in CS-Studio.

kasemir, 2018-10-08:
Corba is no longer needed by the latest code in https://github.com/epics-base/epicsCoreJava/blob/master/pvAccessJava/src/org/epics/pvaccess/impl/remote/IntrospectionRegistry.java, so this should be a matter of changing the dependency to the latest version of pvAccess.

shroffk, 2018-10-10:
I have put out the snapshot release...working on the final release.

shroffk, 2018-10-10:
https://oss.sonatype.org/content/repositories/snapshots/org/epics/epics-pvaccess/5.1.2-SNAPSHOT/

This way you can test that the corba issue is surely resolved.

kasemir, 2018-10-15:
@shroffk , when I put this into the phoebus-target/pom.xml ..

```
<dependency>
      <groupId>org.epics</groupId>
      <artifactId>epics-pvaccess</artifactId>
      <version>5.1.0-20181010.151306-27</version>
</dependency>
```

I get
```
jca-2.3.7-SNAPSHOT.jar
epics-pvaccess-5.1.0-SNAPSHOT.jar 
epics-pvdata-6.1.0-SNAPSHOT.jar 
epics-util-1.0.0-SNAPSHOT.jar
```

What should I use for the vtype?

Browsing https://oss.sonatype.org/content/repositories/snapshots/org/epics the latest vtype I found was https://oss.sonatype.org/content/repositories/snapshots/org/epics/epics-vtype/2.9.1-SNAPSHOT/, which is from 2015.

kasemir, 2018-10-16:
Using the pvaccess snapshot adds dependency on jca-2.3.7-SNAPSHOT.jar, which is a step back from the jca-2.4.1.jar that we've been using, the one that includes CAJ.

kasemir, 2018-10-16:
Is part of the problem that we were already on epics-pvaccess-5.1.1, epics-pvdata-6.1.1, jca-2.4.1, and the suggested 5.1.0-SNAPSHOT may point to files with a new date, but it's overall going back version-wise?
There is a https://oss.sonatype.org/content/repositories/snapshots/org/epics/epics-pvaccess/5.1.2-SNAPSHOT, which suggests going forward in versions, but the files are dated Dec. 2017, i.e. impossible to include the source code updates from June 2018.
Should we give up on maven and just clone the sources, build, declare success?

shroffk, 2018-10-16:
That would be a short term hack.

This is a good opportunity for me to update the release and versioning mechanisms of the epicsCoreJava project too. Maven is behaving as it should...

The issue is not with maven but with the versions on the master branch of the epicsCoreJava project...which we are restructuring and moving from multiple repo's into one. The 5.1.1, 6.1.1, releases happened off the old repo which are now part of epics-rip.

I am waiting on a current PR to close after which I will update all the version numbers and set up new snapshot releases.

kasemir, 2018-10-16:
OK, then I'll stay put until the release.
Here's what we need:

 * [ ] Updated pvAccess library that includes the June 12 removal of Corba dependency (https://github.com/epics-base/epicsCoreJava/commits/master/pvAccessJava/src/org/epics/pvaccess/impl/remote/IntrospectionRegistry.java) 
 * [ ] pvData that goes with it
 * [ ] JCA 2.3.7 or later, the one that includes CAJ
 * [ ] vtypes
 * [ ] epics.util that goes with those vtypes

Once I have that, I can remove the phoebus.vtype, remove some of the phoebus.util, and the Java 11 build should again be able to use pvAccess.

berryma4, 2018-10-16:
Do we still care about Jython? I'm not sure of its support in Java 11.

kasemir, 2018-10-16:
Yes, we need jython for rules, scripts, scan server.
I don't see a problem.
This is Phoebus built & run with Java 11,
running the script-based "Solitaire" demo:

[image]

kasemir, 2018-10-16:
That's with jython-standalone-2.7.1.jar, the latest I can find on https://search.maven.org

kasemir, 2018-10-16:
While I continue to depend on jython, I don't know about continuing support for Javascript.
We went from the external rhino javascript interpreter to the bundled nashorn, which is deprectated in Java 11 and about to be removed.

shroffk, 2018-11-13:
We need to update the instructions for setting up the eclipse IDE for openJDK11 and openJfx

kasemir, 2018-11-13:
> missing javafx libraries
Those libraries are OS-specific. So for windows they need to include
```
javafx-base-11.jar
javafx-base-11-win.jar
javafx-controls-11.jar
javafx-controls-11-win.jar
javafx-fxml-11-win.jar
javafx-graphics-11.jar
javafx-graphics-11-win.jar
javafx-media-11.jar
javafx-media-11-win.jar
javafx-swing-11-win.jar
javafx-web-11-win.jar
```
.. and for mac and linux the respecitive other ones.
The dependencies/phoebus-target/target/lib contains them all.
Right now we list them one-by-one in the .classpath file, which would require 3 different, OS-specific .classpath files.

I tried something like this, but it doesn't work:
```
<?xml version="1.0" encoding="UTF-8"?>
<classpath>
        <classpathentry exported="true" kind="lib" path="target/lib" including="*.jar"/>
        <classpathentry kind="output" path="bin"/>
</classpath>
```

So we could have just instructions, "Add all target/lib/*jar" to the dependencies/phoebus-target/.classpath". Or we could have 3 .classpath-win, .classpath-linux, .classpath-mac with instructions "Rename the appropriate one to just .classpath". Or we could have a script (ant build.xml) that creates the .classpath by listing all  target/lib/*jar

shroffk, 2018-11-14:
> "Add all target/lib/*jar to the dependencies/phoebus-target/.classpath"

This sounds like a bad idea from dependency management point you view...each modules depends on everything.

Maybe /lib/javafx-*.jar would be better

> Or we could have 3 .classpath-win, .classpath-linux, .classpath-mac with instructions "Rename the appropriate one to just .classpath".

This might makes developers unhappy

> Or we could have a script (ant build.xml) that creates the .classpath by listing all target/lib/*jar

Ok..

kasemir, 2018-12-15:
While don't have a perfect solution for the OS-specific dependencies/phoebus-target/..classpath, #502 updates the README.MD to explain that you need to add the javafx*.jar files,
and it offers this commented-out snipped in the .classpath:
```
<!-- On Windows, need to add this:
<classpathentry exported="true" kind="lib" path="target/lib/javafx-base-11-win.jar"/>
<classpathentry exported="true" kind="lib" path="target/lib/javafx-base-11.jar"/>
<classpathentry exported="true" kind="lib" path="target/lib/javafx-controls-11-win.jar"/>
...
-->
```