ControlSystemStudio/phoebus #148: Window Management: Horizontal & Vertical Slit

kasemir, 2018-05-07:
Add option to tab's context menu to create "horizontal" or "vertical" split.

Result in a sash-based left/right respectively top/bottom pane arrangement between users can now move tabs.

Ideally, a "merge" is not needed because closing the last tab in a sash will merge the empty section back into its parent.

kasemir, 2018-05-07:
This means that the code in DockStage.getDockPane() needs to be extended to look for DockPanes not just in the 'Center' of the expected Border Layout. It needs to drill down into the children of the sash panes.

As a side effect, this will allow sites to create site-specific main window layouts, for example with a machine status view in one section of the scene, while the now more flexible DockStage could would still find DockPanes.

pavel-ch, 2018-05-25:
This implementation and optionally "fixed" parameter is a good base. From my experience, it would be very good allow applications like Display Builder to address particular panes from within Action Buttons or Scripts.
For this, I propose to define a "name" parameter of the pane (defined in memento just like "fixed" parameter). This name can be then used as a parameter for Open Display action, and maybe in another applications (like something as "open related display" action from Alarm handler, etc.).
This makes sense in "fixed" pane setups, of course. But from my experience, display runtimes tend to use fixed setups, because of fast operator orientation.

kasemir, 2018-05-25:
The display builder's view of the window system that we have now is pretty close to a web page's view of the browser. From within a page, you can open a new display as a replacement, in a new tab, or in a new window. Just as in a web browser, a display cannot open a new display inside a specific _other_ section, because the existence of such other section would be hard to enforce.
If you want more control over the layout, you can have a large top level display with several embedded screens, and then replace the content of those embedded screens.

willrogers, 2018-05-31:
Note: that's how the _beamlines_ at DLS are using CS-Studio. You already know that the accelerator controls at DLS are using the EDM style, and I believe that this would be well-supported by Phoebus.

kasemir, 2018-05-31:
Tom, I believe what you have in mind is already possible.
Create a window with two horizontal sections:
1) Large 'left' section that shows the 'main' display. Lock that pane. All buttons on the 'main' display are configured to open new panels in a separate tab.
2) Smaller 'right' section, not locked.

When you now press a button in the main section, it opens a new tab.
New tabs are opened in the next available non-locked pane, i.e. the 'right' one.
Users can leave the newly opened tab in the right section, or detach, or move into an already detached second window.

Anyway, I'll keep your request in mind. Still, at the front of my mind is the fact that I never had to explain how to use the web browser to any beam line visitor, while there continue to be complaints on the current Eclipse behavior. The way it 'splits' areas when you move tabs around, the way tabs maximize/minimize, the way 'editors' pop up and move 'views' to the side is not appreciated. Keeping it simple seems not only easier but actually best.

thomascobb, 2018-05-31:
Thanks for that, good to know what is possible. I just realized though that I did not describe the entire situation, so I'll expand for completeness:

[image]

The bottom section is the "Synoptic" containing links to components of the beamline. If you click on one it opens a screen in the main "Component" area. clicking on links in that screen either opens a new tab in the "Component" area, or in the right hand "Details" area.

I think that might be suitably complicated that locking the central pane wouldn't be sufficient.

I agree totally with your comment about web browsers. We have been experimenting with a web GUI for our PandABox, and have a similar paradigm:

[image]

The difference here is that there is no dragging tabs out, re-ordering tabs, just 2 side panes that display content. We have investigated pop-ups to allow those panes to be opened in a new browser window, and that seems to work reasonably well. I guess I could probably do a similar thing in phoebus having a fixed layout with embedded windows that update from what is selected, but it seems a shame to lose the ability to interact with tabs and dynamically resize panes.

pavel-ch, 2018-06-04:
What Tom mentioned is exactly a typical use case of my suggestion.
The implementation could be simple, and fully compatible with existing behavior. If the target pane "name" was not found in the runtime, the display can be opened according to the current rules.
And, moreover, if thinking about window management, do not forget about popups (possibility to open a Display Builder display into a standalone & always on top utility window near the current mouse position). This is a very common and useful feature in OPI software. A good example would be the "Opi Probe", or some other custom control or information dialogs.
I think both suggestions are not too complex for implementation, and add a certain flexibility level to the OPI window management scheme. Both features are common (in a wide range of implementation details) into other OPI/HMI software.

willrogers, 2018-06-18:
We had a discussion about named panes during the meeting last week, and we agreed in principle that this could be supported.

There is one subtlety: what should happen if no 'pane' with the specified name exists? There are two options:

### Ignore the name and create the opi wherever it would have gone anyway 

This is simple, but won't group together screens with that name, when they were intended to be grouped.

### Create a new pane with the specified name

This would allow the groupings to be honoured, but I could imagine a situtation where you are using someone else's screens and it creates a pane where you weren't expecting one.

kasemir, 2018-06-20:
The DockPanes can now be named. Naming as well as splitting, locking, unlocking is possible from the context menu of the tabs for users who are authorized.

Display builder actions to open a display, if the target is a new Tab, can specify a specific pane by name:
[image]

If there is no pane by that name, one is created, but the original idea is to use a layout that contains such a named pane in the desired location. A named pane will not be merged when its last item is closed so that future tabs can again use that same named pane.

This concludes the basic window management/split ticket.