ControlSystemStudio/phoebus #154: Display Builder Editor 'drag' of new widget into display fails on Mac OS X

kasemir, 2018-05-08:
@claudio-rosati noticed that we can no longer drag a new widget from the editor's palette into the display. Clicking a widget in the palette and then 'rubber banding' its desired location works fine, but dragging it fails with 
```
Exception in thread "JavaFX Application Thread" java.lang.IndexOutOfBoundsException: 39, 7
	at javafx.graphics/com.sun.prism.Image$BaseAccessor.getIndex(Image.java:933)
	at javafx.graphics/com.sun.prism.Image$BaseAccessor.getArgb(Image.java:973)
...
	at javafx.graphics/com.sun.prism.Image.getPixels(Image.java:675)
	at javafx.graphics/javafx.scene.image.Image$2.getPixels(Image.java:1219)
	at javafx.swing/javafx.embed.swing.SwingFXUtils.fromFXImage(SwingFXUtils.java:289)
	at org.csstudio.display.builder.editor.util.WidgetTransfer.createDragImage(WidgetTransfer.java:320)
```

Works fine on Linux, the error only happens on Mac OS X.

kasemir, 2018-05-08:
In the debugger, it looks like the icon image is sized 16x16, but SwingFXUtils.fromFXImage somehow treats it as a larger image. Is that related to the presence of `@2x` variants of the images?

claudio-rosati, 2018-05-08:
Yes, it could.
try to move out the @2x and @3x version and see if it works or not.

kasemir, 2018-05-08:
I think the problem is in DragSupport createDragImage:
```
            SwingFXUtils.fromFXImage(image, bbImage);
            g2d.drawImage(bbImage, (int) ( ( width - w ) / 2.0 ), (int) ( ( height - h ) / 2.0 ), null);
...
        final WritableImage dImage = new WritableImage(width, height);
        SwingFXUtils.toFXImage(bImage, dImage);
```

It converts the JFX image into an AWT image and back.
The JFX image is nominally 16x16, but really 32x32. The SwingFXUtils don't know that.
They ask for pixel 17 +, but the nominal JFX image doesn't have that.

Removing the back and forth conversion should fix it.

claudio-rosati, 2018-05-08:
good

kasemir, 2018-05-08:
Ah, OK.
Feel free to update it again, using the default size.
You're on a Mac, so you'll be able to see the original problem.
I'm mostly on Linux which is why I missed it in the first place.

kasemir, 2018-05-09:
Thanks for the update

closed 2018-05-09 by kasemir