ControlSystemStudio/phoebus #174: Phoebus launch application in new window

shroffk, 2018-06-05:
With the integration with cs-studio, it would be useful to be able to have a argument to instruct phoebus to open the resource in a new window

eg
pv://?sim://noise&app=probe&newWindow=true

kasemir, 2018-06-06:
Sure, could add something like `..&target=window|tab&..`.

What worries me is that this creates an unsatisfying intermediate option.

What we have right now:
New tabs open in the existing Phoebus window. That's very predictable. Users can easily locate the new pane, then use in place and close, or detach, or move etc.

The other extreme would be that the new app magically starts in the place that the user had in mind. Like a new window located at the mouse position, or to the right of the RCP view that initiated the launch. This would at least require addition options like `..&target=window&pos=mouse..` or  `..&target=window&xpos=242&ypos=3544&width=123&height=2134&..` . A lot of work not only to implement but also to use. In reality, few external launches will be configured to include detail beyond just `..&target=window`, so the new window pops up anywhere, most likely on the wrong monitor, which is worse than just opening in the existing Phoebus window.

kasemir, 2018-08-30:
Implemented the following:

As before, opens in new tab of existing window:
```
phoebus.sh -resource file:/path/to/display.opi
```

New, opens a new window:
```
phoebus.sh -resource file:/path/to/display.opi?target=window
```

Probably more interesting, opens in a named pane.
Name should be `[a-z_]+`.
If pane doesn't exist, a new window is opened with just that pane.
So when called the first time, it will open a new window.
When then calling again, those will be new tabs in that window:
```
phoebus.sh -resource file:/path/to/display.opi?target=css_pane
```

The CS-Studio RCP integration could thus always call phoebus with `-resource ...&target=integration`

kasemir, 2018-08-31:
Close?