ControlSystemStudio/phoebus #178: License

claudio-rosati, 2018-06-07:
There is no license file in the top project.

After the finding of a colleague, I suggest the following one: [GLWTPL](https://github.com/me-shaon/GLWTPL)... ;-)

kasemir, 2018-06-07:
Most of the java sources refer to the EPL, which is included on the top level of the cs-studio, display builder, https://github.com/kasemir/phoebus-sns, ..
So it would make sense to include a copy of the EPL, like https://github.com/kasemir/phoebus-sns/blob/master/LICENSE . 
But there's also some truth to your suggestion. Incidentally, the topic was yesterday added to the preliminary agenda for next Mon/Tue at the EPICS meeting workgroup meeting.

shroffk, 2020-06-04:
@kasemir 
EPL is the current license, could we close this issue.

closed 2020-06-04 by kasemir