ControlSystemStudio/phoebus #183: dependencies packaging

shroffk, 2018-06-07:
Should the dependencies project be packaged as a jar?
Wouldn't it make more sense as a pom since the goal is to create the lib folder.

kasemir, 2018-06-07:
Yes, the desired outcome is a lib folder that contains all external dependencies.

Maven happens to create a dependencies-0.0.1-SNAPSHOT.jar which we don't need. If maven can skip that,  and only create the target/lib, that would be fine.

shroffk, 2018-06-07:
> Maven happens to create a dependencies-0.0.1-SNAPSHOT.jar which we don't need. If maven can skip that, and only create the target/lib, that would be fine.

I think that should be possible.

kasemir, 2018-11-09:
https://controlssoftware.sns.ornl.gov/css_phoebus offers builds that includes target platforms.
To build without maven, one can download & unzip phoebus-target-linux.zip, phoebus-target-mac.zip,  phoebus-target-win.zip and then build 100% locally with ant.

closed 2022-08-10 by shroffk