ControlSystemStudio/phoebus #188: Building and running Phoebus on pi

shroffk, 2018-06-17:
During the epics meeting 2018 there were suggestions for trying to make cs-studio/phoebus run on pi.
In the past a similar effort was attempted by RAON but the strong dependence on eclipse/swt rather been the reason why this effort was abandoned.

Stewart Cambel made some initial attempts with the following findings
```
I tried to run/install pheobus and got into some minor issues.  I was running on rasbian 9.4 and I found that the default java is 8 and there are some target = 1.9 in the build so I had to upgrade to openjdk9.

The next problems I had were about javafx.blah not being found.  I did see if I could get it working, by adding the maven-javafx plugin... which seemed to work, but then I started to get ant-javafx.jar could not be found errors and I decided to give up! (for the moment) :)
```

shroffk, 2018-06-17:
TODO:

- [ ] install oracle jdk 9/10 on pi
- [ ] ensure inclusion of javafx libraries into the classpath

kasemir, 2018-06-17:
Compare #143 

Java 11 previews, i.e. the JDK and JavaFX, are available for Linux, Mac, Windows on x64 from either Oracle or open JDK. Pi as well as 32 bit in general are no longer offered.

https://blogs.oracle.com/jtc/build-jdk-10-for-your-raspberry-pi-right-on-your-device has some pointers, but I don't see a download.

stuartcampbell, 2018-06-17:
I started to rebuild the openjdk-10 debian unstable packages, but it needs g++7 for some reason.  Giving up for today.  Will try tomorrow with something other than raspbian.

kasemir, 2018-10-04:
JavaFX 11 , https://gluonhq.com/products/javafx/, currently has an ARM SDK. So after moving phoebus up to Java 11 #143, Raspberry Pi might become an option.

shroffk, 2020-06-04:
I have managed to create a version of phoebus for pi.
http://cs-studio.sourceforge.net/nsls2/nsls2.html

It is not regularly updated, we can work on that if the need arises.

closed 2020-06-04 by shroffk