ControlSystemStudio/phoebus #218: Alarm services

shroffk, 2018-07-16:
A project issue ticket for the various pieces of the alarm environment.

- [ ] Alarm state message logger

- [ ] Alarm command message logger

- [ ] Alarm config message logger

kasemir, 2018-07-17:
Will these reports be in a web client, using some existing tools for elastic search where all that's left to do is define some query-type configuration, or will this be Java code & UI running in Phoebus?

shroffk, 2018-07-17:
> Will these reports be in a web client, using some existing tools for elastic search where all that's left to do is define some query-type configuration, or will this be Java code & UI running in Phoebus?

Both,
I will add a simple phoebus UI. I would like a context menu action which brings up the table with the alarm history of a pv.

shroffk, 2019-03-20:
The alarm services have been implemented.
new feature requests should be opened as new issues.

closed 2019-03-20 by shroffk