ControlSystemStudio/phoebus #277: Elastic search data maintenance

kasemir, 2018-08-17:
Describe a way to for basic data maintenance to for example delete logged messages that are older than 3 month.
Might ‘split’ and ‘shrink’ shards, for example create one shard per month and then delete older ones.

es-sns121, 2018-08-17:
There seem to be two options for managing elastic search data by date.

1. **Delete documents older than a specific date using the query APIs.**

This would entail using a chron job and curl to send a "delete by query" message to elastic search (reference 1).

For example, 
```
curl -X POST "localhost:9200/accel_alarms/_delete_by_query" -H 'Content-Type: application/json' -d'
{
    "query":
    {
         "range" : 
          {
               "@timestamp" : 
               {
                    "gte" : "08/01/2018",
                    "lte" : "09/01/2018",
                    "format": "dd/MM/yyyy||yyyy"
               }
          }
    }
}
'
```
This method of deleting the messages using the query APIs allows for the use of single index, however the query operation would likely take a long time.

The favoured option and the one proposed by elastic search developers who are asked this question is the following option (reference 1).

2. **Using time based indices and deleting ones past a certain date with elastic curator.**

Using time based indices would entail creating a new index every week (or day, month, year...) and then deleting the indices as they become outdated. This deletion is managed by a program called curator, part of the elastic stack (reference 5). Having multiple indices does not complicate queries as elastic search handles searches across multiple indices under the hood. It "intelligently" grabs the indices it needs for the given search dates (reference 1).

**References:**

1. https://discuss.elastic.co/t/delete-documents-by-timestamp/92058/11
2. https://www.elastic.co/guide/en/elasticsearch/reference/current/indices-shrink-index.html
3. https://www.elastic.co/guide/en/elasticsearch/reference/current/indices-split-index.html
4. https://www.elastic.co/guide/en/elasticsearch/reference/current/cluster-health.html
5. https://www.elastic.co/guide/en/elasticsearch/client/curator/current/about-features.html

**Future Reading:**

1. https://www.elastic.co/blog/managing-time-based-indices-efficiently

es-sns121, 2018-08-17:
**Curator**

- **Install**
```sudo pip install elasticsearch-curator```
- **config file (YAML)**
```
---
client:
  hosts:
    - localhost
  port: 9200
  use_ssl: False
  ssl_no_validate: False
  http_auth: False
  timeout: 30
  master_only: True
logging:
  loglevel: INFO
  logfile:
  logformat: default
```
   - **actions file (YAML)**
```
---
actions:
  1:
    action: delete_indices
    description: >-
      delete indices older than a month.
    options:
      ignore_empty_list: True
      disable_action: False
    filters:
    - filtertype: pattern
      kind: prefix
      value: accelerator_alarms_
    - filtertype: age
      source: name
      direction: older
      timestring: '%Y.%m.%d'
      unit: days
      unit_count: 30
```
- **Running**
```curator --config /path/to/config.yml /path/to/actions.yml```

es-sns121, 2018-08-17:
To run curator and see the actions it will take without actually completing the actions, you can use the "dry run" feature.

```curator --config /path/to/config.yml /path/to/actions.yml --dry-run ```

Curator will the process the actions file but not actually complete the actions.

es-sns121, 2018-08-17:
**Possible Curator Setup**
- **config file (YAML)**
```
---
client:
  hosts:
    - localhost
  port: 9200
  use_ssl: False
  ssl_no_validate: False
  http_auth: False
  timeout: 30
  master_only: True
logging:
  loglevel: INFO
  logfile:
  logformat: default
```
- **actions file (YAML)**
```
---
actions:
  1:
    action: delete_indices
    description: >-
      delete indices older than three months.
    options:
      ignore_empty_list: True
      disable_action: False
    filters:
    - filtertype: pattern
      kind: prefix
      value: accelerator_alarms_
    - filtertype: age
      source: name
      direction: older
      timestring: '%Y.%m.%d'
      unit: days
      unit_count: 90
 2:
    action: create_index
    description: "Create weekly index"
    options:
     name: '<accelerator_alarms_{now/d +1d}>'
     # ...
     extra_settings:
         settings:
           number_of_shards: 1
           number_of_replicas: 0
         mappings:
           alarm:
             properties:
               APPLICATION-ID:
                 type: text
                 index: not_analyzed
               config:
                 type: keyword
               pv:
                 type: text
                 analyzer: keyword
              severity:
                 type: keyword
              message:
                 type: text
              value:
                 type: text
              time:
                 type: date
                 format: yyyy-MM-dd HH:mm:ss.SSS
              current_severity:
                 type:keyword
              current_message: 
                 type:text
              mode:
                 type: text
```
 - **crontab** 
```
0 0 * * 0 curator --config /path/to/config.yml /path/to/actions.yml
```

Runs every week on Sunday at midnight. Names index 'accelerator_alarms_date' where date is Monday's date.

es-sns121, 2018-08-17:
Testing this 
```
curator --config config.yml actions.yml --dry-run
```
as a dry run resulted in
```
2018-08-17 14:56:15,043 INFO      Preparing Action ID: 1, "delete_indices"
2018-08-17 14:56:15,051 INFO      Trying Action ID: 1, "delete_indices": delete indices older than three months.
2018-08-17 14:56:15,066 INFO      Skipping action "delete_indices" due to empty list: <class 'curator.exceptions.NoIndices'>
2018-08-17 14:56:15,066 INFO      Action ID: 1, "delete_indices" completed.
2018-08-17 14:56:15,066 INFO      Preparing Action ID: 2, "create_index"
2018-08-17 14:56:15,069 INFO      Trying Action ID: 2, "create_index": Create weekly index
2018-08-17 14:56:15,069 INFO      "<accelerator_alarms_{now/d +1d}>" is using Elasticsearch date math.
2018-08-17 14:56:15,069 INFO      DRY-RUN MODE.  No changes will be made.
2018-08-17 14:56:15,070 INFO      DRY-RUN: create_index "<accelerator_alarms_{now/d +1d}>" with arguments: {'mappings': {'alarm': {'properties': {'current_message': {'type': 'text'}, 'current_severity': {'type': 'keyword'}, 'pv': {'type': 'text', 'analyzer': 'keyword'}, 'severity': {'type': 'keyword'}, 'APPLICATION-ID': {'index': 'not_analyzed', 'type': 'text'}, 'value': {'type': 'text'}, 'mode': {'type': 'text'}, 'time': {'type': 'date', 'format': 'yyyy-MM-dd HH:mm:ss.SSS'}, 'message': {'type': 'text'}, 'config': {'type': 'keyword'}}}}, 'settings': {'number_of_replicas': 0, 'number_of_shards': 1}}
2018-08-17 14:56:15,070 INFO      Action ID: 2, "create_index" completed.
2018-08-17 14:56:15,070 INFO      Job completed.
```

shroffk, 2018-08-21:
@es-sns121 
Overall this looks promising.

I have a few questions...outside of Kibana, I create my queries using the java rest client, I am guessing that those queries would now have to be written to include the appropriate weekly created indexes.

shroffk, 2018-08-21:
Also, 
For the alarm logging service...I am guessing it would have to be rewriten too, so that it is synchronized with the cleanup and creation of new indices. Existing streams would have to be destroyed and a new stream would have to created for the new index (after it has been created by the curater)

es-sns121, 2018-08-22:
Based off of the dev response (reference 1 in the original comment) no change to the queries are needed so long as the index pattern matches all of the indexes you are querying over.

es-sns121, 2018-08-22:
According to [this link](https://www.elastic.co/guide/en/elasticsearch/reference/current/multi-index.html) only the document APIs and single index alias APIs do not support querying over multiple indices.

es-sns121, 2018-08-22:
[This link](https://www.elastic.co/guide/en/elasticsearch/reference/current/date-math-index-names.html) also demonstrates using date math to select only specific indices should you want to exclude certain indices from your searches but not aggressively delete outdated indices.

shroffk, 2018-08-22:
Great,
So the search queries just need to be slightly modified to `index=accelerator_alarms_*` , this would be easy.

With regards to my second question, when creating a new entry (I can never get the terminology right)...indexing a new alarm message. 
We would have to use some date calculations to figure out which index to write to, I believe we would have to be specific about this index. It would mean that the logger service would have to be aware of the data management strategy and be synchronized with it - you need to know what is the index name and be sure that it is created and ready.

es-sns121, 2018-08-22:
I believe all the logger would need to know is the current date if there isn't some way to handle this on the elasticsearch side of things. Because any new document that is to be written into an index would only ever go into the latest index. Is that correct?

es-sns121, 2018-08-22:
Never mind, even when writing to the latest index you would still need to do some date math to figure out the current month that the index represents.

es-sns121, 2018-08-22:
However, there does seem to be a way to automatically create an index when writing a document to an index that does not yet exist. And with the use of index [templates](https://www.elastic.co/guide/en/elasticsearch/reference/current/indices-templates.html), the index can be created using the alarms mapping. So, the alarm logger wouldn't need to worry about if the index is created yet and ready or not.

kasemir, 2018-08-22:
Here's my understanding of our current plan:

The alarm logger will get a configuration setting like `index_span_units=M/W/D` and `index_span_amount=2`. When left empty, nothing changes, it writes the same "accelerator_alarms" type of index.

Otherwise it allows configuring an index time span of 1 week or 2 month.
When configured for 2 month, and started on January 4th, it first writes an index "accelerator_alarms_2018_01_01", then 2 months later changes to "accelerator_alarms_2018_03_01". Even will try some helper class and unit test for turning that index span config into good looking index names, i.e. basically rounding down from 'now' to get the currently valid index name until it's time to start the next index.

Elastic, when configured with an index template, will automatically create the index as soon as the alarm logger switches to the next index name. Client tools like Kibana can use wildcards like "accelerator_alarms*" to read from all indices, so neither elastic nor clients need any manual intervention.

The curator deletes older indices. It doesn't care which specific new index the logger is currently writing.
The only problem could occur if  you keep restarting the alarm logger with different index_span configuration settings. In that case, might have to delete existing indices and start over.

kasemir, 2019-01-10:
Considering this more than good enough.

Starting alarm-logger with `-date_span_units D -date_span_value 1` creates one index per day.
Older indices can easily be deleted manually, for example via

`curl -X DELETE "localhost:9200/*_alarms_2018-08-*"`

closed 2019-01-10 by kasemir

shroffk, 2019-01-25:
TODO: move the details of this issue into a readme or help document