ControlSystemStudio/phoebus #286: Context menu on tab: interchanged texts "Split Horizontally / Vertically"

pavel-ch, 2018-08-23:
Minor bug:
In the pane Tab context menu, texts "Split Horizontally / Vertically" are interchanged against icon and functionality.
I do not send a PR because the procedure is too complex for such a small bug.

A suggestion:
for an empty unnamed Pane (just opened by split and considered unwanted later), it should be possible to "close" the pane manually (context menu), instead of the necessity to open or move a content Tab and close the tab or move content back, which is somehow tricky.

kasemir, 2018-08-23:
How are they interchanged?

[image]

"Split Horizontally" creates a left and right section. The icon shows a two-column arrangement with left and right section.

"Split Vertically" creates an upper and lower section. The icon shows a two-row arrangement with an upper and lower section.

I'll look at a context menu for the empty pane to allow closing it.

pavel-ch, 2018-08-23:
Ok I supposed having split line horizontal on "Split Horizontally". Just closing issue.

closed 2018-08-23 by pavel-ch

shroffk, 2018-08-23:
> for an empty unnamed Pane (just opened by split and considered unwanted later), it should be possible to "close" the pane manually (context menu), instead of the necessity to open or move a content Tab and close the tab or move content back, which is somehow tricky.

+1 for this feature

kasemir, 2018-08-23:
Yes, like that, too. It's been implemented.

kasemir, 2018-08-23:
The split icons are some public domain Eclipse icons which show the arrangement plus there's stuff inside each panel. Is that stuff misleading, does it detract from the plain panel arrangement, so they would be more obvious like this?

[image]

kasemir, 2018-08-23:
Even better: Shows 'stuff' in the left respectively top section that will contain the original content of the pane that's about to be split, and an empty right resp. bottom section.

[image]