ControlSystemStudio/phoebus #294: Feature Request: Source Control

DanielCuneo, 2018-08-24:
Is there a way to use source control with a Phoebus / CSS installation ?
I'm guessing that all the configuration is kept in plain text XML files or something similar.

kasemir, 2018-08-25:
Yes ,  config files, settings, mementos, display files can all be put into e.g. git

kasemir, 2018-08-30:
For example, you create a file "settings.ini" as described in the online help, https://phoebus-doc.readthedocs.io/en/latest/preference_properties.html:
```
org.phoebus.pv.ca/addr_list=gateway.my.site.org 127.0.0.1

org.phoebus.framework.workbench/external_apps=Text Editor,txt|dat|py|ini|db|xml|xsl|css|cmd|sh|st|log|out|md,gedit; Image Viewer,png|jpg|gif|jpeg,eog; PDF Viewer,pdf,firefox

org.phoebus.email/mailhost=smtp.my.site.gov

org.phoebus.ui/top_resources=/path/to/my/start.opi, Start Page
```

and create a script that always starts your product with `-settings /path/to/my/settings.ini`.
That `settings.ini`, as well as displays like `start.opi` are all text files that you can (actually should) keep in git.

closed 2018-08-30 by kasemir

DanielCuneo, 2018-08-30:
I was thinking that the .bob files are the most important. Am I misunderstanding ?

kasemir, 2018-08-30:
Right, those are XML as well, and can be kept in git.
Their format is actually such that modifications to a display tend to only change the affected XML, so a "git diff" is often quite readable.