ControlSystemStudio/phoebus #295: master fails to build

DanielCuneo, 2018-08-24:
Cloned master today and it failed to build with maven.

[phoebus_build.log](https://github.com/shroffk/phoebus/files/2320117/phoebus_build.log)

kasemir, 2018-08-25:
Are you using oracle java 9 or 10?

kasemir, 2018-08-27:
Your error log shows `Fatal error compiling: invalid target release: 1.9`. The project README.MD states "Requirements: JDK9 or later". Do you have java 9 or later? What do you get from `java -version`? Even better, check `mvn -version` to see which version of Java maven will invoke when trying to build. A functioning example would look like this:
```
$ mvn -version
Apache Maven 3.3.9 ...
Java version: 9.0.4, vendor: Oracle Corporation
OS name: "linux"...
```

shroffk, 2018-08-27:
Don't forget to set the correct JAVA_HOME
As Kay mentioned you need jdk9 or newer

DanielCuneo, 2018-08-27:
Sorry Java is really new to me...Java versioning is special.
I'll setup the proper Java and try again. Mark as closed.

>mvn -version  
Apache Maven 3.0.5 (Red Hat 3.0.5-17)
Maven home: /usr/share/maven
Java version: 1.8.0_171, vendor: Oracle Corporation
Java home: /opt/jdk1.8.0_171/jre
Default locale: en_US, platform encoding: UTF-8
OS name: "linux", version: "3.10.0-862.3.3.el7.x86_64", arch: "amd64", family: "unix"

kasemir, 2018-08-27:
You have Oracle Java 8.
You need to get Oracle java 9 or 10.
https://www.oracle.com/technetwork/java/javase/downloads/index.html

kasemir, 2018-08-27:
Instead of building it yourself, you can also download the prebuild phoebus-0.0.1.zip from https://ics-web.sns.ornl.gov/css/phoebus/ , unzip and run, but to execute you'd again need Oracle Java 9 or 10.

kasemir, 2018-09-23:
SNS web site moved to https://controlssoftware.sns.ornl.gov/css_phoebus/