ControlSystemStudio/phoebus #303: Test failed in core-util trying to build the whole project.

claudio-rosati, 2018-09-03:
```
------------------------------------------------------------------------
Building core-util 0.0.1-SNAPSHOT
------------------------------------------------------------------------

--- maven-clean-plugin:2.5:clean (default-clean) @ core-util ---
Deleting /Users/claudiorosati/Projects/GitHub4/phoebus/core/util/target

--- maven-resources-plugin:2.6:resources (default-resources) @ core-util ---
Using 'UTF-8' encoding to copy filtered resources.
skip non existing resourceDirectory /Users/claudiorosati/Projects/GitHub4/phoebus/core/util/src/main/resources
Copying 0 resource

--- maven-compiler-plugin:3.6.2:compile (default-compile) @ core-util ---
Changes detected - recompiling the module!
Compiling 62 source files to /Users/claudiorosati/Projects/GitHub4/phoebus/core/util/target/classes

--- maven-resources-plugin:2.6:testResources (default-testResources) @ core-util ---
Using 'UTF-8' encoding to copy filtered resources.
Copying 9 resources
Copying 0 resource

--- maven-compiler-plugin:3.6.2:testCompile (default-testCompile) @ core-util ---
Changes detected - recompiling the module!
Compiling 41 source files to /Users/claudiorosati/Projects/GitHub4/phoebus/core/util/target/test-classes
/Users/claudiorosati/Projects/GitHub4/phoebus/core/util/src/test/java/org/phoebus/util/time/TimeParserTest.java: /Users/claudiorosati/Projects/GitHub4/phoebus/core/util/src/test/java/org/phoebus/util/time/TimeParserTest.java uses or overrides a deprecated API.
/Users/claudiorosati/Projects/GitHub4/phoebus/core/util/src/test/java/org/phoebus/util/time/TimeParserTest.java: Recompile with -Xlint:deprecation for details.
/Users/claudiorosati/Projects/GitHub4/phoebus/core/util/src/test/java/org/phoebus/util/text/CsvParserDebug.java: /Users/claudiorosati/Projects/GitHub4/phoebus/core/util/src/test/java/org/phoebus/util/text/CsvParserDebug.java uses unchecked or unsafe operations.
/Users/claudiorosati/Projects/GitHub4/phoebus/core/util/src/test/java/org/phoebus/util/text/CsvParserDebug.java: Recompile with -Xlint:unchecked for details.

--- maven-surefire-plugin:2.20:test (default-test) @ core-util ---
Surefire report directory: /Users/claudiorosati/Projects/GitHub4/phoebus/core/util/target/surefire-reports

-------------------------------------------------------
 T E S T S
-------------------------------------------------------
Running org.phoebus.util.array.ListLongTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.08 s - in org.phoebus.util.array.ListLongTest
Running org.phoebus.util.array.ListFloatTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.006 s - in org.phoebus.util.array.ListFloatTest
Running org.phoebus.util.array.CollectionTest
Tests run: 6, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.017 s - in org.phoebus.util.array.CollectionTest
Running org.phoebus.util.array.ArrayFloatTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.021 s - in org.phoebus.util.array.ArrayFloatTest
Running org.phoebus.util.array.ListNumbersTest
Tests run: 33, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.098 s - in org.phoebus.util.array.ListNumbersTest
Running org.phoebus.util.array.ArrayBooleanTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 s - in org.phoebus.util.array.ArrayBooleanTest
Running org.phoebus.util.array.CollectionNumbersTest
Tests run: 38, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.017 s - in org.phoebus.util.array.CollectionNumbersTest
Running org.phoebus.util.array.ArrayLongTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.phoebus.util.array.ArrayLongTest
Running org.phoebus.util.array.IteratorTest
Tests run: 6, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.01 s - in org.phoebus.util.array.IteratorTest
Running org.phoebus.util.array.ListIntTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.004 s - in org.phoebus.util.array.ListIntTest
Running org.phoebus.util.array.ListTest
Tests run: 5, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.012 s - in org.phoebus.util.array.ListTest
Running org.phoebus.util.array.BufferIntTest
Tests run: 5, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.004 s - in org.phoebus.util.array.BufferIntTest
Running org.phoebus.util.array.ListByteTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.004 s - in org.phoebus.util.array.ListByteTest
Running org.phoebus.util.array.ListShortTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.006 s - in org.phoebus.util.array.ListShortTest
Running org.phoebus.util.array.ListMathTest
Tests run: 6, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 s - in org.phoebus.util.array.ListMathTest
Running org.phoebus.util.array.ArrayShortTest
Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.phoebus.util.array.ArrayShortTest
Running org.phoebus.util.array.ListDoubleTest
Tests run: 9, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.007 s - in org.phoebus.util.array.ListDoubleTest
[... 81 lines cut ...]
Tests run: 370, Failures: 8, Errors: 0, Skipped: 0

------------------------------------------------------------------------
Reactor Summary:

dependencies ....................................... SUCCESS [  8.319 s]
phoebus (parent) ................................... SUCCESS [  0.003 s]
core ............................................... SUCCESS [  0.003 s]
core-framework ..................................... SUCCESS [ 25.820 s]
core-security ...................................... SUCCESS [  1.202 s]
core-util .......................................... FAILURE [  3.492 s]
core-vtype ......................................... SKIPPED
core-pv ............................................ SKIPPED
core-ui ............................................ SKIPPED
...
service-scan-server ................................ SKIPPED
service-alarm-logger ............................... SKIPPED
------------------------------------------------------------------------
BUILD FAILURE
------------------------------------------------------------------------
Total time: 39.477 s
Finished at: 2018-09-03T10:32:00+02:00
Final Memory: 29M/1024M
------------------------------------------------------------------------
```

kasemir, 2018-09-03:
`IndexNameHelperTest` needs to set timezone to a specific one because the result depends on timezone. It's a helper for creating a for example a new daily index name every midnight, using the local time for midnight.

kasemir, 2018-09-03:
Try with pull request #304

claudio-rosati, 2018-09-03:
Thank you.
I still continue to keep the Swedish locale, so that kind of problems will be catch.

kasemir, 2018-09-23:
Test failure has been fixed.

closed 2018-09-23 by kasemir