ControlSystemStudio/phoebus #305: Problems in building the main phoebus project

claudio-rosati, 2018-09-03:
The following appears at the beginning of the log after starting building the main project:

```
cd /Users/claudiorosati/Projects/GitHub4/phoebus; JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk-10.0.2.jdk/Contents/Home /Applications/NetBeans/netbeans/java/maven/bin/mvn --errors -e clean install
Error stacktraces are turned on.
Scanning for projects...

Some problems were encountered while building the effective model for org.phoebus:core-pv:jar:0.0.1-SNAPSHOT
'dependencies.dependency.systemPath' for org.epics:jca:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/jca-2.4.1.jar will be unresolvable by dependent projects @ org.phoebus:core-pv:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/core/pv/pom.xml, line 33, column 19
'dependencies.dependency.systemPath' for org.epics:epics-pvdata:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/epics-pvdata-6.1.1.jar will be unresolvable by dependent projects @ org.phoebus:core-pv:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/core/pv/pom.xml, line 40, column 19
'dependencies.dependency.systemPath' for org.epics:epics-pvaccess:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/epics-pvaccess-5.1.1.jar will be unresolvable by dependent projects @ org.phoebus:core-pv:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/core/pv/pom.xml, line 47, column 19

Some problems were encountered while building the effective model for org.phoebus:app-databrowser:jar:0.0.1-SNAPSHOT
'dependencies.dependency.systemPath' for org.tallison:jmatio:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/jmatio-1.2.jar will be unresolvable by dependent projects @ org.phoebus:app-databrowser:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/app/databrowser/pom.xml, line 28, column 19

Some problems were encountered while building the effective model for org.phoebus:app-display-runtime:jar:0.0.1-SNAPSHOT
'dependencies.dependency.systemPath' for org.epics:jca:jar should not point at files within the project directory, ${basedir}/../../../dependencies/target/lib/jca-2.4.1.jar will be unresolvable by dependent projects @ org.phoebus:app-display-runtime:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/app/display/runtime/pom.xml, line 32, column 19
'dependencies.dependency.systemPath' for org.epics:epics-pvdata:jar should not point at files within the project directory, ${basedir}/../../../dependencies/target/lib/epics-pvdata-6.1.1.jar will be unresolvable by dependent projects @ org.phoebus:app-display-runtime:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/app/display/runtime/pom.xml, line 39, column 19
'dependencies.dependency.systemPath' for org.epics:epics-pvaccess:jar should not point at files within the project directory, ${basedir}/../../../dependencies/target/lib/epics-pvaccess-5.1.1.jar will be unresolvable by dependent projects @ org.phoebus:app-display-runtime:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/app/display/runtime/pom.xml, line 46, column 19
'dependencies.dependency.systemPath' for org.python:jython-standalone:jar should not point at files within the project directory, ${basedir}/../../../dependencies/target/lib/jython-standalone-2.7.1.jar will be unresolvable by dependent projects @ org.phoebus:app-display-runtime:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/app/display/runtime/pom.xml, line 53, column 19

Some problems were encountered while building the effective model for org.phoebus:product:jar:0.0.1-SNAPSHOT
'build.plugins.plugin.version' for org.apache.maven.plugins:maven-jar-plugin is missing. @ org.phoebus:product:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/phoebus-product/pom.xml, line 151, column 15

Some problems were encountered while building the effective model for org.phoebus:service-alarm-server:jar:0.0.1-SNAPSHOT
'dependencies.dependency.systemPath' for org.epics:jca:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/jca-2.4.1.jar will be unresolvable by dependent projects @ org.phoebus:service-alarm-server:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/services/alarm-server/pom.xml, line 36, column 19
'dependencies.dependency.systemPath' for org.epics:epics-pvdata:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/epics-pvdata-6.1.1.jar will be unresolvable by dependent projects @ org.phoebus:service-alarm-server:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/services/alarm-server/pom.xml, line 43, column 19
'dependencies.dependency.systemPath' for org.epics:epics-pvaccess:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/epics-pvaccess-5.1.1.jar will be unresolvable by dependent projects @ org.phoebus:service-alarm-server:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/services/alarm-server/pom.xml, line 50, column 19
'build.plugins.plugin.version' for org.apache.maven.plugins:maven-jar-plugin is missing. @ org.phoebus:service-alarm-server:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/services/alarm-server/pom.xml, line 111, column 15

Some problems were encountered while building the effective model for org.phoebus:service-archive-engine:jar:0.0.1-SNAPSHOT
'dependencies.dependency.systemPath' for org.epics:jca:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/jca-2.4.1.jar will be unresolvable by dependent projects @ org.phoebus:service-archive-engine:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/services/archive-engine/pom.xml, line 94, column 19
'dependencies.dependency.systemPath' for org.epics:epics-pvdata:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/epics-pvdata-6.1.1.jar will be unresolvable by dependent projects @ org.phoebus:service-archive-engine:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/services/archive-engine/pom.xml, line 101, column 19
'dependencies.dependency.systemPath' for org.epics:epics-pvaccess:jar should not point at files within the project directory, ${basedir}/../../dependencies/target/lib/epics-pvaccess-5.1.1.jar will be unresolvable by dependent projects @ org.phoebus:service-archive-engine:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/services/archive-engine/pom.xml, line 108, column 19
'build.plugins.plugin.version' for org.apache.maven.plugins:maven-jar-plugin is missing. @ org.phoebus:service-archive-engine:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/services/archive-engine/pom.xml, line 154, column 15

Some problems were encountered while building the effective model for org.phoebus:service-scan-server:jar:0.0.1-SNAPSHOT
'build.plugins.plugin.version' for org.apache.maven.plugins:maven-jar-plugin is missing. @ org.phoebus:service-scan-server:[unknown-version], /Users/claudiorosati/Projects/GitHub4/phoebus/services/scan-server/pom.xml, line 100, column 15

It is highly recommended to fix these problems because they threaten the stability of your build.

For this reason, future Maven versions might no longer support building such malformed projects.

------------------------------------------------------------------------
Reactor Build Order:
...
```

claudio-rosati, 2018-09-03:
Thank you for the explanation. 

Now I also understand why even if the build was successful and I can run the product from a terminal window, I can't from Netbeans where the products project (alone) fails to build because of missing dependencies. 

I'll investigate if those warnings can be suppressed, and I'll try to use the NetBeans project configuration to point in the right places (probably a files like the `.classpath` will be created, or an existing one updated).

kasemir, 2018-09-05:
I think we're mixing two related issues:
1) The original use of systemPath and resulting warnings back when everything was available in maven
2) How to add new jars that are not in maven

On 1:
So far all dependencies were described in the dependencies/pom.xml, fetched from maven central, copied into dependencies/target/lib. Or they could have been copied into dependencies/target/lib by some other means.
I tried to use the content of dependencies/target/lib as the target platform. For ant, eclipse, .. not a problem. For Maven, that meant using systemPath. Could just scrap that idea, since I was the only one to do that anyway. Most of the poms don't use systempath but the normal maven coordinates. Hopefully they're duplicated in dependencies/pom.xml, so dependencies/target/lib shows the same as what's in .m2/repository. We'll find out once maven our builds have different results.

On 2:
If you have an external jar that can't be fetched from maven central, you could place it in dependencies/target/lib via wget or some other mechanism. Everything but maven can then use it, and maven could use systempath for this because after all it's not in maven.
Or as you describe, install it in the local maven repo. If we want to avoid systempath warnings for case 1, then maybe it's best to also avoid them for case 2, so yes, maven then requires a pre-build step to install the plain jars into the local maven repo.

In the end, I'd still prefer to have just one folder that's like a "target platform".
For example, this:
 *  dependencies/get_nonmaven.sh Script that uses wget etc. to fetch jars that are not in maven central, installs them into local maven repo
 * dependencies/pom.xml: Lists all, fetching them from local repo or maven central
 * dependencies/target/lib: The target platform with everything regardless of origin

claudio-rosati, 2018-09-06:
Yes I think that fetching and then [installing local maven artifacts](https://maven.apache.org/guides/mini/guide-3rd-party-jars-local.html) should be the right solution.

What Is not clear to me is if the install goal of the maven-install-plugin can somehow used from the pom itself at the very beginning.

shroffk, 2018-09-06:
I have a solution which I have tried to implement

Under dependencies there are now 2 modules

1. install-jars
This takes existing jars and installs them in the local maven repo.
This can be expanded to wget jars etc...

2. phoebus-target
This is like the previous dependencies project.
It lists all the dependencies. Now this can also include all the non maven dependencies which we have installed with the previous step.
It then copies all the jars to a phoebus-target/target/lib folder which can be used by ant and other builds.

We don't need system dependencies anymore

shroffk, 2018-09-06:
It would also mean that we could package all jar dependencies with install-jars and thus they would be available in both phoebus-target/lib and in the target maven repo

kasemir, 2018-09-23:
`systempath` entries have been removed.

closed 2018-09-23 by kasemir