ControlSystemStudio/phoebus #310: Problems launching phoebus product

claudio-rosati, 2018-09-06:
When I start phoebus-product I have plenty of the following channel finder-related exception (both launching from NetBeans and from terminal):

```
CI0011766:target claudiorosati$ java -jar product-0.0.1-SNAPSHOT.jar 
2018-09-06 14:03:32 INFO [org.phoebus.product.Launcher] Phoebus (PID 31949)
2018-09-06 14:03:34 WARNING [org.phoebus.framework.workbench] external_apps missing name, ext, command for '
2018-09-06 14:03:34 CONFIG [com.sun.jersey.spi.service.ServiceFinder] Running in a non-OSGi environment
2018-09-06 14:03:34 CONFIG [com.sun.jersey.core.spi.component.ProviderServices] Searching for providers that implement: interface com.sun.jersey.client.proxy.ViewProxyProvider
2018-09-06 14:03:34 CONFIG [com.sun.jersey.core.spi.component.ProviderServices] Searching for providers that implement: interface com.sun.jersey.spi.inject.InjectableProvider
2018-09-06 14:03:34 CONFIG [com.sun.jersey.core.spi.component.ProviderServices]     Provider found: class com.sun.jersey.core.impl.provider.xml.SAXParserContextProvider
2018-09-06 14:03:34 CONFIG [com.sun.jersey.core.spi.component.ProviderServices]     Provider found: class com.sun.jersey.core.impl.provider.xml.XMLStreamReaderContextProvider
2018-09-06 14:03:34 CONFIG [com.sun.jersey.core.spi.component.ProviderServices]     Provider found: class com.sun.jersey.core.impl.provider.xml.DocumentBuilderFactoryProvider
2018-09-06 14:03:34 CONFIG [com.sun.jersey.core.spi.component.ProviderServices]     Provider found: class com.sun.jersey.core.impl.provider.xml.TransformerFactoryProvider
2018-09-06 14:03:34 CONFIG [com.sun.jersey.core.spi.component.ProviderServices]     Provider found: class com.sun.jersey.multipart.impl.MultiPartConfigProvider
2018-09-06 14:03:34 FINE [com.sun.jersey.core.reflection.ReflectionHelper] Unable to load class javax.annotation.PostConstruct using the supplied class loader jdk.internal.loader.ClassLoaders$AppClassLoader.
java.lang.ClassNotFoundException: javax.annotation.PostConstruct
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:582)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:190)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:374)
	at com.sun.jersey.core.reflection.ReflectionHelper$2.run(ReflectionHelper.java:221)
	at com.sun.jersey.core.reflection.ReflectionHelper$2.run(ReflectionHelper.java:216)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.jersey.core.spi.component.ComponentConstructor.getPostConstructMethods(ComponentConstructor.java:128)
	at com.sun.jersey.core.spi.component.ComponentConstructor.<init>(ComponentConstructor.java:124)
	at com.sun.jersey.core.spi.component.ProviderFactory.__getComponentProvider(ProviderFactory.java:165)
	at com.sun.jersey.core.spi.component.ProviderFactory.getComponentProvider(ProviderFactory.java:137)
	at com.sun.jersey.core.spi.component.ProviderServices.getComponent(ProviderServices.java:283)
	at com.sun.jersey.core.spi.component.ProviderServices.getProvidersAndServices(ProviderServices.java:208)
	at com.sun.jersey.core.spi.factory.InjectableProviderFactory.configure(InjectableProviderFactory.java:106)
	at com.sun.jersey.api.client.Client.init(Client.java:264)
	at com.sun.jersey.api.client.Client.access$000(Client.java:119)
	at com.sun.jersey.api.client.Client$1.f(Client.java:192)
	at com.sun.jersey.api.client.Client$1.f(Client.java:188)
	at com.sun.jersey.spi.inject.Errors.processWithErrors(Errors.java:193)
	at com.sun.jersey.api.client.Client.<init>(Client.java:188)
	at com.sun.jersey.api.client.Client.<init>(Client.java:171)
	at com.sun.jersey.api.client.Client.create(Client.java:683)
	at org.phoebus.channelfinder.ChannelFinderClientImpl.<init>(ChannelFinderClientImpl.java:278)
	at org.phoebus.channelfinder.ChannelFinderClientImpl$CFCBuilder.create(ChannelFinderClientImpl.java:264)
	at org.phoebus.channelfinder.ChannelFinderService.getClient(ChannelFinderService.java:49)
	at org.phoebus.channel.views.ChannelTableApp.start(ChannelTableApp.java:27)
	at org.phoebus.ui.application.PhoebusApplication.startApplications(PhoebusApplication.java:985)
	at org.phoebus.ui.application.PhoebusApplication.backgroundStartup(PhoebusApplication.java:197)
	at org.phoebus.ui.application.PhoebusApplication.lambda$start$0(PhoebusApplication.java:181)
	at org.phoebus.framework.jobs.Job.execute(Job.java:50)
	at org.phoebus.framework.jobs.JobManager.execute(JobManager.java:49)
	at org.phoebus.framework.jobs.JobManager.lambda$schedule$1(JobManager.java:40)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1135)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:844)
...
<a lot of them>
```

I'm using the following java on my MacOS X 10.13.6

```
java version "10.0.2" 2018-07-17
Java(TM) SE Runtime Environment 18.3 (build 10.0.2+13)
Java HotSpot(TM) 64-Bit Server VM 18.3 (build 10.0.2+13, mixed mode)
```

kasemir, 2018-09-06:
Duplicate of #160

kasemir, 2018-09-06:
See also #311 : If you include it in your product (as done by the default maven build), there's no obvious way to configure it so that it's _not_ trying to connect. As a result, you keep getting error messages whenever you type a PV.

shroffk, 2018-09-06:
The exception comes from the jersey library...
You can change the logging.properties to be at a level above FINE

kasemir, 2018-09-06:
Good idea. I'll leave the jersey settings to you because I'm not sure which ones you need to see and which are superfluous, but yes, please add to https://github.com/shroffk/phoebus/blob/master/phoebus-product/src/main/resources/logging.properties when you find log levels that are best moved up.

shroffk, 2018-09-06:
```
com.sun.jersey = WARNING
org.glassfish.jersey = WARNING
```
Tried both the above within eclipse...but can't get it to work properly.
@claudio-rosati can you try to add the above lines to logging.properties and see if it helps?

kasemir, 2018-09-06:
Need to add ".level":
```
com.sun.jersey.level = WARNING
org.glassfish.jersey.level = WARNING
```

claudio-rosati, 2018-09-07:
Yes, putting the 2 lines above will solve the problem.
Kunal, will you do the pull request or I have to?

shroffk, 2018-09-07:
@kasemir thanks...that did the trick 
We only need the con.sun.jersey setting, I will make a PR.

kasemir, 2018-09-23:
Closing: Duplicate of #160, #311, and superfluous warnings have been disabled.

closed 2018-09-23 by kasemir