ControlSystemStudio/phoebus #314: Testing of phoebus applications

shroffk, 2018-09-07:
The phoebus testing framework needs to be updated to support the following test types.
1. UnitTest : the test class should end with Test
2. IntegrationTestIT : the integration tests should ent with IT
3. ExampleDemo : standalone ui example applications should be places in the test folder with their class name ending with Demo
4. ExampleUI : run user interface tests using testFX

kasemir, 2018-09-07:
Basic info on how TestFX:  https://github.com/es-sns121/testfx-demo

Actual test in phoebus: https://github.com/shroffk/phoebus/blob/master/app/logbook/ui/src/test/java/org/phoebus/logbook/ui/write/TestListSelectionDialog.java

@es-sns121 : For now we'll name those UI tests "...UI", so TestListSelectionDialog will be renamed.

@shroffk : Need a way to selectively run tests, for example incantations of maven with `-D ...` to execute unit tests, or UI tests, or IT

shroffk, 2018-09-07:
The maven sure fire plugin would be the best starting point.
https://maven.apache.org/surefire/maven-surefire-plugin/examples/skipping-tests.html

kasemir, 2018-09-27:
Yes

closed 2018-09-27 by kasemir