ControlSystemStudio/phoebus #327: 404 for POM resource

DanielCuneo, 2018-09-18:
Building master: commit 4c6944ee23e68ac6e3391028847c4e4f974fe924
Build fails b/c of 404 for the following resource:

`
[INFO] Building app-alarm-model 0.0.1-SNAPSHOT                          [39/55]
[INFO] --------------------------------[ jar ]---------------------------------
Downloading from sonatype-nexus-snapshots: https://oss.sonatype.org/content/repositories/snapshots/com/fasterxml/jackson/core/jackson-annotations/2.9.0/jackson-annotations-2.9.0.jar
`

The 2.9 version is not there, but 2.10 is:
https://oss.sonatype.org/content/repositories/snapshots/com/fasterxml/jackson/core/jackson-annotations/2.10.0-SNAPSHOT/jackson-annotations-2.10.0-20180816.040648-1.jar

shroffk, 2018-09-18:
@DanielCuneo I tried to clean up (deleted my maven repo too) and rebuilt the dependencies

It was able to successfully resolve jackson-annotations. However it was resolved to 2.9.4
```
[INFO] ---------------------< org.phoebus:phoebus-target >---------------------
[INFO] Building phoebus-target 0.0.1-SNAPSHOT                             [3/3]
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] Downloading from : https://oss.sonatype.org/content/repositories/snapshots/com/fasterxml/jackson/core/jackson-annotations/2.9.4/jackson-annotations-2.9.4.pom
[INFO] Downloading from : https://repo.maven.apache.org/maven2/com/fasterxml/jackson/core/jackson-annotations/2.9.4/jackson-annotations-2.9.4.pom
[INFO] Downloaded from : https://repo.maven.apache.org/maven2/com/fasterxml/jackson/core/jackson-annotations/2.9.4/jackson-annotations-2.9.4.pom (2.6 kB at 13 kB/s)
```

Could you pull and re-build master

DanielCuneo, 2018-09-19:
I just did a git pull of master:

`[ERROR] Failed to execute goal on project app-alarm-model: Could not resolve dependencies for project org.phoebus:app-alarm-model:jar:0.0.1-SNAPSHOT: Could not transfer artifact com.fasterxml.jackson.core:jackson-annotations:jar:2.9.0 from/to sonatype-nexus-snapshots (https://oss.sonatype.org/content/repositories/snapshots): Connect to oss.sonatype.org:443 [oss.sonatype.org/34.198.109.105, oss.sonatype.org/52.22.108.61, oss.sonatype.org/52.6.213.150] failed: Connection timed out (Connection timed out) -> [Help 1]
`

That's interesting to me that your jackson-annotations resolved to 2.9.4. When I navigated to  the URL I only saw 2.10.x. I'm still really new to Maven / Java so that might not mean much.

shroffk, 2018-09-19:
Hi @DanielCuneo 
It takes time with any build system...no worries :)

Just a few basic questions...are you trying to build all of phoebus or just the alarm service?
Can you try to turn on debugging 
`mvn clean verify -e
`

DanielCuneo, 2018-09-19:
I'm building the entire project: `mvn clean install` from within master.
Running `mvn clean verify -e` now and will report back. 

Sanity check the commit #:
`
git log  | head
commit 726ac2f7b00e865c21905b90b2a471481ac77037
Merge: f896b4d 4478595
Author: es-sns121 <smithej@ornl.gov>
Date:   Wed Sep 19 13:15:51 2018 -0400
`

Cheers.

shroffk, 2018-09-19:
Ok, 
Also, are you running for behind a firewall??

kasemir, 2018-09-19:
Maven downloads things from various places, and more often than not that fails.

In theory, once you built the dependencies (cd dependencies; mvn clean install) you should have all the outside dependencies, and from then on you can build with ant, no more network connection needed.

If maven fails to assemble the dependencies, you can alternatively download them from the SNS site,
https://controlssoftware.sns.ornl.gov/css_phoebus/ , specifically 
https://controlssoftware.sns.ornl.gov/css_phoebus/nightly/phoebus-target.zip

DanielCuneo, 2018-09-20:
BUILD SUCCESS, such sweet ascii 

I was building on a machine with access to the outside, but I think I did ssh into the wrong host.
It's working now.

Yesterday I absolutely was building on the correct host.
Cheers Kunal.

closed 2018-09-20 by shroffk

willrogers, 2018-09-20:
I agree with Kunal. Our problems were much more Eclipse than Maven.

kasemir, 2018-09-20:
I agree that it works most of the time.
Still, see Qs above as to why it's trying for version 2.9 instead of 2.10 or 2.9.4.
With a ZIP of jars, there's no uncertainty.
But don't worry, you can use maven.

shroffk, 2018-09-20:
> EDIT: I think it would be good if there were some release tags, even if they are Alpha. We are starting to build a system around Phoebus and it would be a huge sanity saver down the line when something goes wonky.

Ok, I think we can have a alpha tagge release on the 3rd or Oct.

kasemir, 2018-09-20:
What's special about Oct. 3?

shroffk, 2018-09-20:
The monthly cs-studio meeting