ControlSystemStudio/phoebus #331: pv write fails in phoebus

shroffk, 2018-09-20:
Reported by Dr. Bastian Löher (b.loeher@gsi.de)

Detailed information:

- phoebus commit 70a914b65f087f60987e14f747401fd58a1fabfe
- gateway commit eb331cd6ab45d02265096289a7bea00df4ec77b0
- EPICS base: 3.16.1 (for IOC and gateway)

- phoebus should read and write PVs published by an EPICS gateway

- I need to set EPICS_CA_ADDR_LIST=<gateway>, so my settings.ini file 
  contains:
    org.phoebus.pv.ca/addr_list=127.0.0.1 <gateway>
  Without this, I cannot even read the PVs, as expected.

- caget/caput of these values works without a problem

- cainfo shows (as an example):

$> EPICS_CA_ADDR_LIST=<gateway> cainfo testAPD:scope1:ChannelAThreshold
testAPD:scope1:ChannelAThreshol
    State:            connected
    Host:             <gateway>:41922
    Access:           read, write
    Native data type: DBF_ENUM
    Request type:     DBR_ENUM
    Element count:    1

- phoebus can read all PVs
- phoebus does not determine PVs to be writable (widgets are disabled)

- trying PV probe and writing a value directly yields the following 
  messages:

----

2018-09-20 06:21:19 WARNING [org.phoebus.applications.probe] Cannot write '-100' to PV testAPD:scope1:ChannelAThreshold
gov.aps.jca.CAException: No write access rights granted.
        at com.cosylab.epics.caj.CAJChannel.put(CAJChannel.java:1006)
        at com.cosylab.epics.caj.CAJChannel.put(CAJChannel.java:867)
        at gov.aps.jca.Channel.put(Channel.java:365)
        at org.phoebus.pv.ca.JCA_PV.performWrite(JCA_PV.java:454)
        at org.phoebus.pv.ca.JCA_PV.write(JCA_PV.java:422)
        at org.phoebus.applications.probe.view.ProbeController.lambda$initialize$0(ProbeController.java:101)
        at javafx.base/com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
        at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
        at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
        at javafx.base/com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at javafx.base/com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
        at javafx.base/com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
        at javafx.base/javafx.event.Event.fireEvent(Event.java:198)
        at javafx.graphics/javafx.scene.Scene$KeyHandler.process(Scene.java:4083)
        at javafx.graphics/javafx.scene.Scene$KeyHandler.access$1600(Scene.java:4029)
        at javafx.graphics/javafx.scene.Scene.processKeyEvent(Scene.java:2146)
        at javafx.graphics/javafx.scene.Scene$ScenePeerListener.keyEvent(Scene.java:2620)
        at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler$KeyEventNotification.run(GlassViewEventHandler.java:217)
        at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler$KeyEventNotification.run(GlassViewEventHandler.java:149)
        at java.base/java.security.AccessController.doPrivileged(Native Method)
        at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleKeyEvent$1(GlassViewEventHandler.java:248)
        at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
        at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler.handleKeyEvent(GlassViewEventHandler.java:247)
        at javafx.graphics/com.sun.glass.ui.View.handleKeyEvent(View.java:547)
        at javafx.graphics/com.sun.glass.ui.View.notifyKey(View.java:971)
        at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
        at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:277)
        at java.base/java.lang.Thread.run(Thread.java:844)

----

- It would be nice, if PV probe would also show the R/W rights and the 
  PV data type in the metadata field.

shroffk, 2018-09-20:
+1 for the feature of having probe display write permissions.

kasemir, 2018-09-20:
I'll add an indication to Probe when the PV is readonly.

kasemir, 2018-09-20:
OK, probe now handles write access.

Back to the original issue:

In Phoebus, you don't have write access. Widgets are disabled. Probe wasn't disabled, but when you actually write, you get the  gov.aps.jca.CAException: No write access rights granted. So everything agrees there, from higher level UI down to the basic CA library.

From caget/caput/cainfo, however, you think you should be able to write.

--> Looks like those use different connection settings.

You have `org.phoebus.pv.ca/addr_list=127.0.0.1` and 
`cainfo testAPD:scope1:ChannelAThreshold .. Host: :41922`

What is host 41922? Is that the same as localhost?
Or is the CA client connecting somehow via a local subnet broadcast?

Set both of these:
```
# Set to the IP of your gateway that allows write access
org.phoebus.pv.ca/addr_list= IP OF YOUR GATEWAY
# Set to false to absolutely use the addr_list and not get other entries magically added
org.phoebus.pv.ca/auto_addr_list=false
```

bl0x, 2018-09-20:
Ah, copy and paste ate some of my text.
I originally wrote in the mail:

What I have in my settings.ini:
```
org.phoebus.pv.ca/addr_list=127.0.0.1 gateway_host
```

What cainfo tells:

```
$> EPICS_CA_ADDR_LIST=gateway_host cainfo testAPD:scope1:ChannelAThreshold
testAPD:scope1:ChannelAThreshold
    State:            connected
    Host:             gateway_host:41922
    Access:           read, write
    Native data type: DBF_ENUM
    Request type:     DBR_ENUM
    Element count:    1
```

bl0x, 2018-09-20:
> Set both of these:
> 
> ```
> # Set to the IP of your gateway that allows write access
> org.phoebus.pv.ca/addr_list= IP OF YOUR GATEWAY
> # Set to false to absolutely use the addr_list and not get other entries magically added
> org.phoebus.pv.ca/auto_addr_list=false
> ```

This I added to my settings.ini.

I also checked out the newest commit: 0be2e5684d61cd174f34a3e1c84696344ffeb0b2
and tried again after compiling with ```mvn clean package```.

The result is, that I cannot change the value in the PV probe anymore.

kasemir, 2018-09-20:
OK, makes a little more sense now.
Still, we appear to have two different paths to the PVs here, one that can write, and one that can't.
Set `org.phoebus.pv.ca/auto_addr_list=false` to that it's ONLY using the addr_list, and not silently add all the local interfaces.

From the CA server side, write access is based on
a) The host name
b) The user name

The host name tends to be what `hostname` shows. The user is the currently logged in user.

Connect to the IOC once via 'camonitor' from the command line, and once via probe.
Each time, check the output on the IOC when you type "casr 1".
It should show from which host and as what user the client connects.

Could it be that

bl0x, 2018-09-20:
I actually have both options set in my settings.ini, like so:
```
org.phoebus.pv.ca/addr_list=landgw01
org.phoebus.pv.ca/auto_addr_list=false
```

The gateway hostname is ```landgw01```.
The user and the host are both in the access list of the EPICS gateway.

Issuing ```casr 1``` on the IOC shell, I see the following:

```
epics> casr 1
Channel Access Server V4.13
1 client connected:
    TCP client at 192.168.1.1:59964 'landgw01':
        User 'land', V4.13, Priority = 0, 51 Channels
CAS-TCP server on 0.0.0.0:5064 with
    CAS-UDP name server on 0.0.0.0:5064
Sending CAS-beacons to 1 address:
    192.168.1.255:5065
```

So, this is the gateway making the connection to the IOC.
The gateway is running as a different user (```land``` in this case) as the client (camonitor or PV probe). The IOC itself does not impose any access security. What should matter are the access rights imposed by the gateway.

kasemir, 2018-09-20:
Well, no idea how to check on the gateway what it receives as a client connection.
For some reason your gateway doesn't seem to see the 'cainfo' and the poebus client the same way.

Triple-check that you run both 'cainfo' and poebus as the same user.
On the gateway, add both the simple host name and the fully qualified one, like "landgw01" as well as "landgw01.my.institute.de".

Could be that the CAJ client sends the full name, while cainfo only sends the short name, and the gateway simply compares the string.

bl0x, 2018-09-20:
Ok, found the issue.
And you've also posted the solution now.
I missed the fully qualified name of the client host in the access security of the gateway.

One can get the connection report from the gateway by writing to the ```gateway:report1Flag``` PV.
This gave for the PV in question:

testAPD:scope1:ChannelAThreshold event rate =  0.01
  land@lxir123 (read/write access)
  bloeher@lxir123.gsi.de (read only access)

The second one was the PV probe connection.

Solution: After adding the fully qualified domain name to the access security in the gateway, in addition to the simple host name, the connection from both ```cainfo``` and PV probe works.

Thanks!

bl0x, 2018-09-20:
The issue can be closed now.

A note in the documentation could indicate, that if one wants to use phoebus together with an EPICS gateway, one should make sure to register the FQDN in the gateway configuration.

kasemir, 2018-09-20:
Well, it's not really a phoebus issue but more a generic EPICS issue.
Some vxWorks IOCs, when acting as client, will send the short or the long name.
You can configure your Linux host's `hostname` with long or short.

So all CA security files, no matter if plain IOC or gateway, always best list both full and short host names, no matter if the client is another IOC, CAJ, gateway, ...

closed 2018-09-20 by kasemir