ControlSystemStudio/phoebus #341: Try to compile phoebus on Ubuntu 16.04.5 LTS

hjunkes, 2018-09-23:
Can't compile on Ubuntu 16.04.5 LTS with java-9-openjdk-amd64 or jdk-10.0.2. 

[h1@earth phoebus (master)]$ ant clean dist
Buildfile: /home/h1/PHOEBUS/phoebus/build.xml

clean:

clean:
   [delete] Deleting directory /home/h1/PHOEBUS/phoebus/core/framework/target

clean:

...

clean:

core:

core-framework:
    [mkdir] Created dir: /home/h1/PHOEBUS/phoebus/core/framework/target/classes
    [javac] Compiling 62 source files to /home/h1/PHOEBUS/phoebus/core/framework/target/classes
    [javac] /home/h1/PHOEBUS/phoebus/core/framework/src/main/java/org/phoebus/framework/spi/MenuEntry.java:7: error: package javafx.scene.image does not exist
    [javac] import javafx.scene.image.Image;
    [javac]                          ^
    [javac] /home/h1/PHOEBUS/phoebus/core/framework/src/main/java/org/phoebus/framework/spi/MenuEntry.java:43: error: cannot find symbol
    [javac]     public default Image getIcon()
    [javac]                    ^
    [javac]   symbol:   class Image
    [javac]   location: interface MenuEntry
    [javac] /home/h1/PHOEBUS/phoebus/core/framework/src/main/java/org/phoebus/framework/spi/ToolbarEntry.java:3: error: package javafx.scene.image does not exist
    [javac] import javafx.scene.image.Image;
    [javac]                          ^
    [javac] /home/h1/PHOEBUS/phoebus/core/framework/src/main/java/org/phoebus/framework/spi/ToolbarEntry.java:24: error: cannot find symbol
    [javac]     public default Image getIcon()
    [javac]                    ^
    [javac]   symbol:   class Image
    [javac]   location: interface ToolbarEntry
    [javac] Note: Some input files use or override a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] Note: Some input files use unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 4 errors

BUILD FAILED
/home/h1/PHOEBUS/phoebus/build.xml:65: The following error occurred while executing this line:
/home/h1/PHOEBUS/phoebus/core/framework/build.xml:6: Compile failed; see the compiler error output for details.

Total time: 3 seconds

[h1@earth phoebus (master)]$ mvn clean install
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/usr/share/maven/lib/guice.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[INFO] Scanning for projects...
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO] 
[INFO] phoebus (parent)
[INFO] dependencies
[INFO] install-jars
[INFO] phoebus-target
[INFO] core
[INFO] core-framework
[INFO] core-security
[INFO] core-util
[INFO] core-vtype
[INFO] core-pv
[INFO] core-ui
[... 377 lines cut ...]
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 5.669 s
[INFO] Finished at: 2018-09-23T14:44:46+02:00
[INFO] Final Memory: 27M/97M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.6.2:compile (default-compile) on project core-framework: Compilation failure: Compilation failure:
[ERROR] /home/h1/PHOEBUS/phoebus/core/framework/src/main/java/org/phoebus/framework/spi/ToolbarEntry.java:[3,26] package javafx.scene.image does not exist
[ERROR] /home/h1/PHOEBUS/phoebus/core/framework/src/main/java/org/phoebus/framework/spi/ToolbarEntry.java:[24,20] cannot find symbol
[ERROR] symbol:   class Image
[ERROR] location: interface org.phoebus.framework.spi.ToolbarEntry
[ERROR] /home/h1/PHOEBUS/phoebus/core/framework/src/main/java/org/phoebus/framework/spi/MenuEntry.java:[7,26] package javafx.scene.image does not exist
[ERROR] /home/h1/PHOEBUS/phoebus/core/framework/src/main/java/org/phoebus/framework/spi/MenuEntry.java:[43,20] cannot find symbol
[ERROR] symbol:   class Image
[ERROR] location: interface org.phoebus.framework.spi.MenuEntry
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :core-framework
[h1@earth phoebus (master)]$

kasemir, 2018-09-23:
You need a Java with JavaFX.

The easiest way is the Oracle 9 or 10 JDK.

For openJDK, see  https://github.com/shroffk/phoebus/issues/143 for an example of using the openJDK 11 early acccess preview.
I'm not sure about OpenJDK 9/10 and JavaFX. Try something like
`apt-get install -y openjfx` to add JavaFX.

kasemir, 2018-09-23:
The longer story is that we use JavaFX for the UI.
Oracle Java 8, 9, 10 SDKs include JavaFX.
For OpenJDK, JavaFX is a separate install.

kasemir, 2018-09-23:
https://stackoverflow.com/questions/18547362/javafx-and-openjdk

hjunkes, 2018-09-23:
Danke Kay fuer die schnelle Antwort.

I have now installed oracle java 10 for the sake of simplicity.
With the installation of openjfx I did not get any further. Maybe I got myself confused
 when trying to get channelfinder to work with glassfish and Co.

mvn clean install
...
...
[INFO] service-archive-engine ............................. SUCCESS [  1.029 s]
[INFO] service-scan-server ................................ SUCCESS [  2.130 s]
[INFO] service-alarm-logger ............................... SUCCESS [ 15.203 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 05:10 min
[INFO] Finished at: 2018-09-23T15:28:46+02:00
[INFO] Final Memory: 57M/194M
[INFO] ------------------------------------------------------------------------

Looks great.

For licensing reasons we are currently trying to get rid of all applications that use Oracle Java (whether JDK or JRE) .

Heinz

closed 2018-09-23 by hjunkes

kasemir, 2018-09-23:
Good to hear that it works!

With Java 11, it looks like JavaFX will become a dependency just like the others, installed by either building the `dependencies` package with maven, or by downloading a `phoebus-target.zip` from a site like  https://controlssoftware.sns.ornl.gov/css_phoebus/ .
This means we no longer expect JavaFX to be included in the JDK. Plain OpenJDK would then work as well as Oracle JDK, and which JDK you choose would then simply be based on what you think runs better or fits your licensing persuasion.