ControlSystemStudio/phoebus #347: Image tab view

shroffk, 2018-09-27:
It seemed like we were each developing our own widget for displaying a group of images..
I recently tried to replace my implementation with `org.phoebus.ui.javafx.ImagesTab` Now the log entry table looks as follows.

[image]

However,
1. It seems like I can't get the images to be of the correct resolution
2. I would like us to create a read only view where the buttons are not visible
3. In my old image title pane, I had added a double click action which would pop up then image in a new  view.

kasemir, 2018-09-27:
Sounds good.

1 - The preview indeed seems blurry. As I move the split pane between the 'large' preview on the left and the list of 'small' images on the right, the 'small' images can actually be made large, and then they seem crisp. But the 'preview' is blurry, even though it should have the exact same image. 

2 - Read only should be simple, just a `boolean read_only` constructor option.

3 - Open preview, would that open a new dialog that shows the image? Or are you talking about a DockItem?

shroffk, 2018-09-27:
> 2 - Read only should be simple, just a boolean read_only constructor option.

Ok, I can do this

> 3 - Open preview, would that open a new dialog that shows the image? Or are you talking about a DockItem?

I had implemented it as a new dialog.

kasemir, 2018-09-28:
Dialog is good; go ahead with a `read_only` option; I'm puzzled about the blurry preview

shroffk, 2019-03-20:
I think I solved the blurry image by playing with size definitions of the image. In the current version this problem should not exist.

@kasemir I think this ticket can be closed.

closed 2019-03-20 by kasemir