ControlSystemStudio/phoebus #353: 3D Display "WARNING: System can't support ConditionalFeature.SCENE3D"

kasemir, 2018-10-08:
When I ran the new 3D Viewer on Linux inside a Virtual Box VM, it would not show anything but instead keep printing
```
WARNING [javafx.scene....]: System can't support ConditionalFeature.SCENE3D
```

The same phoebus product worked fine on the Linux host as well as on a Mac.

But when running on Centos inside a VM, it showed that warning with both Oracle JDK 10 (build from master) and OpenJDK 11 (build from the java11 branch).
Same problem when connecting via ssh from the Linux host where everything was fine into the Centos VM, so the problem doesn't seem to be the X server.

What fixed it:
Adding `-Dprism.forceGPU=true`.

Other suggestions found online:
`-Dprism.verbose=true` generates a lot of output.
`-Dcom.sun.javafx.experimental.embedded.3d=true` and `-Dprism.glDepthSize=24` might help with 3D related problems as well, but didn't make a difference for me.

kasemir, 2018-10-10:
More detail on the good:

 * Same behaviour for Oracle JDK 9, Oracle 10, OpenJDK 11
 * For Linux, found no difference for `-Djdk.gtk.version=2` vs. `-Djdk.gtk.version=3`
 * No problem on Mac
 * OK when directly running on RHEL 7.5
 * Running on Centos 7.5 inside VirtualBox on RHEL 7.5 requires `-Dprism.forceGPU=true`
 * Running on Centos 7.5 inside VirtualBox when connected via ssh from RHEL 7.5 requires `-Dprism.forceGPU=true`

.. the unknown ..
 * Haven't tried Windows

.. the bad:
 * Running on a remote RHEL 7.5 connected via ssh from RHEL 7.5 results in Prism-ES2 Error and then `System can't support ConditionalFeature.SCENE3D`
 * Same when running on a remote RHEL 7.5 via ThinLic (remote desktop in web browser)

kasemir, 2018-10-10:
Just found that in the 'bad' case the issue must be related to ssh. All's fine when sitting at that remote computer, but not when connected via ssh to the same remote computer.

es-sns121, 2018-10-10:
Enabling iGLX may resolve this. It seems that when JavaFX can't get a hardware accelerated renderer, it defaults to its 2D pipeline to make it easier on the software renderer.

I've tried enabling iglx by editing /etc/X11/xorg.conf and enabling it, but have had no success as of yet. 
I've tried running glxgears on a remote server and it fails, so I don't believe I've got iglx enabled.

closed 2019-06-20 by kasemir