ControlSystemStudio/phoebus #377: W10 Build, on Startup Fails to Copy prefs.tmp to prefs.xml

DanielCuneo, 2018-10-24:
I built Phoebus on Windows 10 and have this problem:

`2018-10-24 14:38:05 WARNING [java.util.prefs] Couldn't flush user prefs: 
java.util.prefs.BackingStoreException:

Can't rename 
C:\Users\username\.phoebus\.userPrefs\org\phoebus\pv\ca\prefs.tmp to 
C:\Users\username\.phoebus\.userPrefs\org\phoebus\pv\ca\prefs.xml
`

Copying the paths manually gets things working again.

kasemir, 2018-10-25:
We use a copy of the default `java.util.prefs.FileSystemPreferences` for Unix on _all_ platforms to have better control over where and how the prefs are stored. The code is in `org.phoebus.framework.preferences.FileSystemPreferences`.
Maybe there's something in there that doesn't work on Windows?
@shroffk , you're developing on Windows, do you see this error?

shroffk, 2018-10-25:
@DanielALS While I have not seen that exact error. I have seen a similar "can't rename" error when dealing with files which had been included in the eclipse workspace (just as links) for some reason eclipse had help one to the file and was not allowing a rename.

Do you see this error while running phoebus from within eclipse?
I am guessing this error occours on shutdown...is that correct?

kasemir, 2018-10-25:
> I am guessing this error occours on shutdown...is that correct?

Right, the complete stack trace might help.
There are several places where we set & flush preferences, for example in the DialogHelper where Claudio added code to preserve the last location. But those result in a warning and then move on:
```
                try
                {
                    prefs.flush();
                }
                catch (BackingStoreException ex)
                {
                    logger.log(Level.WARNING, "Unable to flush preferences", ex);
                }
```

So there are really two things here:
1) That specific "Can't rename" error. With the default phoebus home of $HOME/.phoebus, that's unexpected. A user should always be able to rename his/her own files.
2) A problem saving the preferences should only result in a warning and not impact the program execution beyond that.

shroffk, 2018-10-25:
> That specific "Can't rename" error. With the default phoebus home of $HOME/.phoebus, that's unexpected. A user should always be able to rename his/her own files.

I think this is due to the file being locked by another application...which is why it would be useful to know if this issue is with a standalong instance of phoebus or when running phoebus from the IDE.

> A problem saving the preferences should only result in a warning and not impact the program execution beyond that.  

+1

DanielALS, 2019-02-02:
Has any work been done on this ?
I've found another related error mode that is more serious.

When I edit the settings.ini file to comment out an addr_list IP and add a new one, there's no effect.
Here's an email excerpt from the person whom I'm supporting:

> I am trying to test the RF GUI in Phoebus.  I changed the addr_list in 
settings.ini to llrf.als.lbl.gov instead of the test one we used to 
have.  The pref.xml is updated as

<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE map SYSTEM "http://java.sun.com/dtd/preferences.dtd">
<map MAP_XML_VERSION="1.0">
   <entry key="addr_list" value="llrf.als.lbl.gov"/>
</map>

> This shows that settings.ini was applied. However, in Phoebus, my 
display for BL01 Allen Bradley test program still works (I can 
open/close valve in my office).

I then commented out the line for addr_list in settings.ini. Same thing 
happened.

kasemir, 2019-02-02:
No work has been done because nobody has been able to reproduce the issue. Any news on the ASAP stack trace from Oct of last year?

As for the address list:
The values from the settings file are placed in the preferences.
So strictly speaking you only need to run with `-settings my_settings.ini` once, and it then remembers them.
Deleting a line from my_settings.ini thus has no effect for the current user.
When you change some setting to a different value, you should from then on get the different values, though.
When you add a new IP, do the connections to the new IP work, and the concern is that you can still _also_ connect to some previous IP? Maybe there's another path to the channels on the previous IP? For example, did you leave `auto_addr_list=true`, and the default broadcast is still finding PVs that you no longer expect to find?

DanielALS, 2019-02-04:
> When you add a new IP, do the connections to the new IP work, and the concern is that you can still also connect to some previous IP? 

Yes that is the case.

> Maybe there's another path to the channels on the previous IP? For example, did you leave auto_addr_list=true, and the default broadcast is still finding PVs that you no longer expect to find?

Just checked and the auto option is commented out.
I'll trying explicitly setting it to false. I bet that solves the issue.

**Stack Trace for rename error**
I don't have direct access to that machine, so I asked the user to copy and paste the stack trace.
This is what I got :\

`2019-02-04 14:06:35 WARNING [java.util.prefs] Couldn't flush user prefs: java.ut
il.prefs.BackingStoreException: Can't rename C:\Users\username\.phoebus\.userPref
s\org\phoebus\pv\ca\prefs.tmp to C:\Users\username\.phoebus\.userPrefs\org\phoebu
s\pv\ca\prefs.xml`

pavel-ch, 2019-03-18:
Okay, seems I have to enter this discussion, too. I am denveloping on Windows, 2 machines (Win 7 and Win 10), getting the same exception (since beginnings of Phoebus app):

> 2019-03-18 16:14:41 WARNING [java.util.prefs] Couldn't flush user prefs: java.util.prefs.BackingStoreException: Can't rename C:\Users\xxx\.phoebus\.userPrefs\org\csstudio\display\builder\model\prefs.tmp to C:\Users\xxx\.phoebus\.userPrefs\org\csstudio\display\builder\model\prefs.xml

The exception comes from returning `false`:
```
private native boolean rename0(File f1, File f2)
```
called from
```
WinNTFileSystem.class.rename(File f1, File f2)
```
The destination file is NOT locked by another application (can be successfully renamed from another application). After that, the rename operation succeedes once, then fails again.
Apparently, the version of Rename operation does not work on Windows if destination file exists.
See also Javadoc:

> boolean java.io.File.renameTo(File dest)
> 
> 
> Renames the file denoted by this abstract pathname. 
> 
> Many aspects of the behavior of this method are inherently platform-dependent: The rename operation might not be able to move a file from one filesystem to another, it might not be atomic, and it might not succeed if a file with the destination abstract pathname already exists. The return value should always be checked to make sure that the rename operation was successful. 
> 
> Note that the java.nio.file.Files class defines the move method to move or rename a file in a platform independent manner.
> 

I would suggest to use the above mentioned java.nio.file.Files move method. But do not know other possible efects, so I do not send a PR yet. I ask for your opinions.

kasemir, 2019-03-18:
Thanks, that sheds some light.

We copied the `java.util.prefs.FileSystemPreferences` from the Unix version, because we wanted to use the same type of files on all platforms, instead of using those files on Unix, but the registry on Windows etc.

Looks like that doesn't work 100% because the 'rename' option as used in there doesn't work on Windows as it did on Unix (and Mac OS).
Makes sense to replace that.
Note that the file browser would have the same problem. org.phoebus.applications.filebrowser.RenameAction also currently calls File.renameTo.

pavel-ch, 2019-03-18:
For me, this works:
```
    /**
     * Attempt to write back prefsCache to the backing store. If the attempt
     * succeeds, lastSyncTime will be updated (the new value will correspond exactly
     * to the data thust written back, as we hold the file lock, which prevents a
     * concurrent write. If the attempt fails, a BackingStoreException is thrown and
     * both the backing store (prefsFile) and lastSyncTime will be unaffected by
     * this call. This call will NEVER leave prefsFile in a corrupt state.
     */
    private void writeBackCache() throws BackingStoreException {
        try {
            AccessController.doPrivileged(new PrivilegedExceptionAction() {
                @Override
                public Object run() throws BackingStoreException {
                    try {
                        if (!dir.exists() && !dir.mkdirs())
                            throw new BackingStoreException(dir + " create failed.");
                        FileOutputStream fos = new FileOutputStream(tmpFile);
                        FilePreferencesXmlSupport.exportMap(fos, prefsCache);
                        fos.close();
                        Files.move(tmpFile.toPath(), prefsFile.toPath(), StandardCopyOption.REPLACE_EXISTING);
                    } catch (Exception e) {
                        throw new BackingStoreException(e);
                    }
                    return null;
                }
            });
        } catch (PrivilegedActionException e) {
            throw (BackingStoreException) e.getException();
        }
    }

```
The operation, again, is not necessarily atomic, but probably is being on the same filesystem.
Should I send PR?

kasemir, 2019-03-18:
Yes, please create PR.

pavel-ch, 2019-04-03:
I would suggest to close this issue as this is resolved by the PR #580

closed 2019-04-03 by kasemir