ControlSystemStudio/phoebus #408: Acknowledged alarm does not leave active alarm list

johnsinsheimer, 2018-11-07:
A PV configured with the alarm server to latch became an active alarm, and then returned to the OK state.  Clicking acknowledge does not remove it from the active alarm list.  The PV also shows as OK under the alarm tree.

[image]

shroffk, 2018-11-07:
I think you meant to say that other alarms with current severity != OK were successfully acknowledged. Only alarms with current severity == OK could not be acknowledged.

In the above scenario, these "OK" alarms being acknowledged should completely disappear from the table.

johnsinsheimer, 2018-11-07:
Correct Kunal, sorry for not being clearer.  I haven't checked if it is because the current severity is == OK or if it is because the current severity is just different than the latched alarm severity.

shroffk, 2018-11-07:
Ah, it would be great if you could test that scenario....my feeling is that for some reason the alarms are not being cleared when acknowledged.

kasemir, 2018-11-07:
When that happens, we need to understand if the alarm server didn't get the PV update, or the UI didn't get the state update.

Check in the alarm server console what state the PV has in there. Try 'help', 'ls -alarms', 'ls /Accelerator/sub/xx/the_pv'.

Try the https://github.com/shroffk/phoebus/blob/master/app/alarm/examples/monitor_topic.sh script to see what the last state update for that PV was:

`monitor_topic.sh AlarmState | fgrep NameOfPV`

shroffk, 2018-11-07:
Oh the script looks cool

I was looking at > ls -active to check if the active alarm was being cleared. 
but this looks more easy

kasemir, 2018-11-08:
Well, they're all important in combination, since I'm not sure what's happening:

Is the alarm server not seeing the current PV state? (ls .. at the alarm server console)

Did the alarm server not send the appropriate updates to the *State topic? (monitor_topic.sh, actually for both XXXState and plain XXX in case the config was change around the same time)

Did the UI not receive & handle the update? (Errors in the application log?)

kasemir, 2018-11-09:
ChenXin reported a similar issue:

A PV is in alarm. Alarm server console `ls` reflects it correctly:
```
1 : /Accelerator/Linac/A/A:ai1 [A:ai1] - connected, VDouble[8.0, MAJOR(HIHI_ALARM), 2018/11/08 19:02:01.895] - annunciating - latching - Current: MAJOR/HIHI_ALARM (8.0), 2018-11-08 19:02:01.895, Alarm: MAJOR/HIHI_ALARM (8.0), 2018-11-08 19:02:01.895
```

`monitor_topic.sh AcceleratorState` also shows the matching last (and only) update:
```
CreateTime:1541674921895: /Accelerator/Linac/A/A:ai1: {"severity":"MAJOR","message":"HIHI_ALARM","value":"8.0","time":{"seconds":1541674921,"nano":895058896},"current_severity":"MAJOR","current_message":"HIHI_ALARM"}
```

But a newly opened alarm tree shows "OK".

Looking into it.

kasemir, 2018-11-09:
Branch alarm_updates_408 enables more logging. Setting `org.phoebus.applications.alarm.level = FINE` in the logging properties will show state updates seen by the UI.

kasemir, 2018-11-15:
ChenXin identified the scenario that resulted in a UI inconsistent with the actual alarm state.

Some PV was in alarm, got then deleted, and added back in.
On the 'XXX' topic used for configuration, there were messages like this:

```
Time 1 PV description="Original description"
Time 5 PV null  (deleted)
Time 10 PV description="Added back in"
```

On the 'XXXState' topic used for alarm state updates, there were messages like this:
```
Time 2 PV state=MINOR
Time 11 PV state=MINOR
```

The UI subscribes to both streams. The message content makes it obvious whether we're looking at a config or state update, affecting separate parts of an alarm tree item. When a null message is received, that item is removed from the alarm tree. When receiving a message for a previously unknown item, it's created with some initial assumption for the alarm state, expecting to soon receive the actual alarm state.

While messages from within each topic are received in the correct order, messages from two different topics can arrive out-of-order like this:

```
...
Time 11 PV state=MINOR
Time 5 PV null  (deleted)
Time 10 PV description="Added back in"
```

So the item updated to the last known alarm state, was then deleted, and added back in, loosing the last known alarm state.

This is only a problem when items are deleted and added back in.
Without the deletion, the out-of-order merging from the two topics wouldn't matter:

```
...
Time 11 PV state=MINOR
Time 10 PV description="Added back in"
```

This would update the alarm state, then the description. Both alarm state and description end up at the last known correct value.

Merging the topic streams based on time would add infinite delays:
Having received message A1 from topic A, you could not react until you also receive message Bx from topic B to check how they correlate in time.
But since the issue is limited to item deletion, the fix may be to just check when receiving null (deletion) messages: If we already saw a state update with a later time stamp, ignore the deletion.

kasemir, 2018-12-10:
ChenXin found that the same issue of receiving an outdated deletion message from the config topic applies to any node, not just leaf nodes (PVs). Erroneously deleting an intermediate node will delete all PVs in the subtree, so need to track time of last state update for any client node.

closed 2019-09-03 by kasemir