ControlSystemStudio/phoebus #427: Systemd issue with alarm server service

tanviash, 2018-11-14:
I'm trying to use systemd to start alarm-server service but it seems to endlessly restarting the service. It's because the command shell that opens gets no argument from systemd. Is there a way to start alarm-server service without the interactive shell?

Here is my systemd file.
deployer@phoebus-alarm-dev:~$ sudo cat /lib/systemd/system/phoebus-alarm-server.service 
#####  /lib/systemd/system/phoebus-alarm-server.service #####

[Unit]
Description=Phoebus Alarm Server

[Service]
Type=simple
#Environment=NODE_ENV=production
#WorkingDirectory=/home/deployer/phoebus

# ExecStart must be absolute path to executable
EnvironmentFile=/etc/phoebus_alarm_server/alarm_server.conf
ExecStart=/usr/bin/java -Duser.home=/var/lib/phoebus-alarm-server -jar /usr/share/phoebus-alarm-server/service-alarm-server-0.0.1-SNAPSHOT.jar -server $ARG1 -config $ARG2
StandardOutput=journal+console
User=phoebus-alarm-server
#Group=nogroup

Restart=always

SyslogIdentifier=phoebus-alarm-server
#SyslogFacility=local0
#SyslogLevel=info

[Install]
WantedBy=multi-user.target

Here is the journalctl log.

deployer@phoebus-alarm-dev:~$ sudo journalctl -u phoebus-alarm-server -f
-- Logs begin at Tue 2018-11-13 15:54:48 EST. --
Nov 14 11:18:17 phoebus-alarm-dev systemd[1]: phoebus-alarm-server.service: Failed with result 'exit-code'.
Nov 14 11:18:22 phoebus-alarm-dev systemd[1]: Started Phoebus Alarm Server.
Nov 14 11:18:23 phoebus-alarm-dev phoebus-alarm-server[9595]: 2018-11-14 11:18:23 INFO [org.phoebus.applications.alarm] Alarm Server (PID 9595)
Nov 14 11:18:23 phoebus-alarm-dev phoebus-alarm-server[9595]: 2018-11-14 11:18:23 INFO [org.phoebus.applications.alarm] Server: kafkatest01.nscl.msu.edu:9092
Nov 14 11:18:23 phoebus-alarm-dev phoebus-alarm-server[9595]: 2018-11-14 11:18:23 INFO [org.phoebus.applications.alarm] Config: Accelerator
Nov 14 11:18:23 phoebus-alarm-dev phoebus-alarm-server[9595]: 2018-11-14 11:18:23 INFO [org.phoebus.applications.alarm] Fetching past alarm states...
Nov 14 11:18:23 phoebus-alarm-dev phoebus-alarm-server[9595]: SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
Nov 14 11:18:23 phoebus-alarm-dev phoebus-alarm-server[9595]: SLF4J: Defaulting to no-operation (NOP) logger implementation
Nov 14 11:18:23 phoebus-alarm-dev phoebus-alarm-server[9595]: SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Nov 14 11:18:24 phoebus-alarm-dev phoebus-alarm-server[9595]: 2018-11-14 11:18:24 INFO [org.phoebus.applications.alarm] Reading from start of AcceleratorState
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]: 2018-11-14 11:18:36 INFO [org.phoebus.applications.alarm] Start handling alarms
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]: 2018-11-14 11:18:36 INFO [org.phoebus.applications.alarm] Not using any heartbeat PV
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]: Commands:
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]: Note: '.' and '..' will be interpreted as the current directory and the parent directory respectively.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]: Spaces within a path do not need to be quoted.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         ls               - List all alarm tree items in the current directory.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         ls -disconnected - List all the disconnected PVs in the entire alarm tree.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         ls -disabled     - List all the disabled PVs in the entire alarm tree.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         ls -all          - List all alarm tree PVs in the entire alarm tree.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         ls -active       - .. which are in active alarm.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         ls -alarm        - .. alarm, active or acknowledged.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         ls dir           - List all alarm tree items in the specified directory contained in the current directory.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         ls /path/to/dir  - List all alarm tree items in the specified directory at the specified path.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         cd               - Change to the root directory.
Nov 14 11:18:36 phoebus-alarm-dev phoebus-alarm-server[9595]:         cd dir           - Change to the specified directory contained in the current directory.
Nov 14 11:18:36 phoebus-alarm-dev systemd[1]: phoebus-alarm-server.service: Service hold-off time over, scheduling restart.
Nov 14 11:18:36 phoebus-alarm-dev systemd[1]: Stopped Phoebus Alarm Server.
Nov 14 11:18:36 phoebus-alarm-dev systemd[1]: Started Phoebus Alarm Server.

kasemir, 2018-11-14:
Here's what I use:
```
# File /etc/systemd/system/phoebus_alarm_server.service
#
# After changes
#   sudo systemctl daemon-reload
#
# Start/status/stop
#   sudo systemctl start phoebus_alarm_server.service
#   sudo systemctl status phoebus_alarm_server.service
#   sudo systemctl stop phoebus_alarm_server.service
#
# Start when computer boots:
#   sudo systemctl enable phoebus_alarm_server.service
#
# Log file:
#   less +GF /home/controls/css/phoebus-alarm-server/console.log
#
# Console:
#   telnet localhost 4609

[Unit]
Description=Phoebus Alarm Server
After=kafka.service

[Service]
User=ky9
Group=slowcontrols_developers
Environment=JAVA_HOME=/home/controls/css/jdk-10
Environment=SERVER=/home/controls/css/phoebus-alarm-server
Environment=PORT=4609
Environment=CONFIG=IHC
Environment=PATH=/home/controls/css/jdk-10/bin:/usr/bin
ExecStart=/usr/bin/procServ --foreground --noautorestart --name alarm-server --chdir ${SERVER} --logfile ${SERVER}/console.log ${PORT} ./alarm-server.sh -settings /home/controls/css/phoebus.ini -settings /home/controls/ih/css/phoebus.ini -config ${CONFIG}
# Systemd stops alarm server by killing the process

[Install]
WantedBy=multi-user.target
```

kasemir, 2018-11-14:
The main thing seems to be that I run it under procServ for several reasons:
Output is logged, you can 'telnet' to its console to issue commands.
We run all IOCs under procServ as well.

tanviash, 2018-11-14:
Thanks. This helps a lot.

closed 2018-11-14 by tanviash

kasemir, 2018-11-14:
Those service scripts are actually included in the app/alarm/examples folder, see *.service in https://github.com/shroffk/phoebus/tree/master/app/alarm/examples