ControlSystemStudio/phoebus #429: Phoebus app: sim://ramp or sim://sine not going into alarm state

tanviash, 2018-11-14:
I created some simulated pvs under my alarm tree. However these sim pvs don't seem to go into alarm state. On probing the pv through phoebus GUI, I can see its value varying between its range and alaso going into alarm range, but that doesn't change the status of pv to 'alarm'.

This is what my alarm server shell show about this pv.
/Accelerator/root > ls -all     
  1 : /Accelerator/root/sim:\/\/ramp [sim://ramp] - connected, VDouble[3.0, NONE(NONE) - None, 2018-11-14T20:43:26.338384Z] - annunciating - latching - Current: OK/None (3.0), 2018-11-14 15:43:26.338, Alarm: OK/OK (-5.0), 2018-11-14 15:27:43.338
[image]

kasemir, 2018-11-14:
It's a side effect of the recent update to EPICS 7.0.2 vtype.

If you can still see them go into alarm in the GUI, that means you're not building the GUI off the latest master branch. I see no alarms on sim://sine, sim://ramp.  Only sim://intermittent disconnects periodically as designed and that results in an alarm.

I'll have to re-enable alarms on sim:// PVs.

kasemir, 2018-11-14:
Found the culprit, the sim:// PVs always used `Alarm.none()`.

tanviash, 2018-11-14:
Thanks Kay. I'll close the issue now.

tanviash, 2018-11-14:
Hi Kay, After merging your recent changes 08817d7, I still do not see the sim pvs going into alarm state. It's just the sim://intermittent that goes into alarm state.

kasemir, 2018-11-14:
Check if you rebuilt.

I just downloaded the latest build from https://controlssoftware.sns.ornl.gov/css_phoebus/ and sure enough the sim://sine PV again has alarm state:

[image]

kasemir, 2018-11-14:
[image]

shroffk, 2018-11-15:
I can confirm...I checked out the master and the alarms are available.

tanviash, 2018-11-15:
Thanks. It works. Closing this issue.

closed 2018-11-15 by tanviash